PhotonVision/photonvision #1149: The raspberry pi image is asking for a password upon boot , which password do i use ?

pkonnoth, 2024-01-12:
**Describe the bug**
A clear and concise description of what the bug is and what the expected behavior should have been.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64):
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch):
 - PhotonVision Version:
 - Browser (with Version) (Chrome, Edge, Firefox, etc.):
 - Camera(s) Used:

**Additional context**
Add any other context about the problem here.

BytingBulldogs3539, 2024-01-13:
It is the default raspberry pi login username "pi" password "raspberry"

mcm001, 2024-01-13:
But why do you need to log in in the first place?

closed 2024-02-11 by gerth2