PhotonVision/photonvision #1151: Linux / Debian 11: libwpiutiljni.so could not be loaded from path

Dannyzen, 2024-01-12:
**Describe the bug**

Hi there, I'm trying to get photonvision running on an arm64 debian 11 device. My current thinking is that the PATH variable is not being picked up by photonvision.jar based on the below (though welcome an alternative diagnosis):

```
java.io.IOException: /root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so could not be loaded from path
        attempted to load for platform /linux/arm64/
Last Load Error: 
/root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so: /root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so: file too short

        at edu.wpi.first.util.CombinedRuntimeLoader.loadLibrary(CombinedRuntimeLoader.java:158)
        at edu.wpi.first.util.CombinedRuntimeLoader.loadLibraries(CombinedRuntimeLoader.java:194)
        at org.photonvision.common.util.TestUtils.loadLibraries(TestUtils.java:57)
        at org.photonvision.Main.main(Main.java:307)
[2024-01-13 04:27:07] [General - Main] [INFO] Native libraries loaded.
java.io.IOException: /root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so could not be loaded from path
        attempted to load for platform /linux/arm64/
Last Load Error: 
/root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so: /root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535/libwpiutiljni.so: file too short

        at edu.wpi.first.util.CombinedRuntimeLoader.loadLibrary(CombinedRuntimeLoader.java:158)
        at edu.wpi.first.util.CombinedRuntimeLoader.loadLibraries(CombinedRuntimeLoader.java:194)
        at org.photonvision.common.util.TestUtils.loadLibraries(TestUtils.java:57)
        at org.photonvision.mrcal.MrCalJNILoader.forceLoad(MrCalJNILoader.java:29)
        at org.photonvision.Main.main(Main.java:322)
[2024-01-13 04:27:07] [Camera - MrCalJNILoader] [ERROR] Couldn't load shared object mrcal_jni: /tmp/libmrcal_jni.so: libopencv_calib3d.so.4.8: cannot open shared object file: No such file or directory
java.lang.UnsatisfiedLinkError: /tmp/libmrcal_jni.so: libopencv_calib3d.so.4.8: cannot open shared object file: No such file or directory
        at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
        at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
        at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
        at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2394)
        at java.base/java.lang.Runtime.load0(Runtime.java:755)
        at java.base/java.lang.System.load(System.java:1953)
        at org.photonvision.jni.PhotonJNICommon.forceLoad(PhotonJNICommon.java:61)
        at org.photonvision.mrcal.MrCalJNILoader.forceLoad(MrCalJNILoader.java:49)
        at org.photonvision.Main.main(Main.java:322)
```

What's confusing is that if I explicitly set the path variable in a shell, prior to run to: 
`PATH=PATH:**/root/.wpilib/nativecache/linux/arm64/1cc8e0c0c9df1063c6f3789cda757535**/lib:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games`

I still get the same issue when invoking
`sudo java -jar photonvision.jar`

[journal_ctl_example.txt](https://github.com/PhotonVision/photonvision/files/13923837/journal_ctl_example.txt)
[photonvision-2024-1-13_04-30-17.log](https://github.com/PhotonVision/photonvision/files/13923842/photonvision-2024-1-13_04-30-17.log)

mcm001, 2024-01-12:
Iirc someone else saw the same error message -- maybe search through GitHub issues and discord message history to see if you can find that?

Dannyzen, 2024-01-23:
Tried installation today and receive the same result on 
```

Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy

```

Why is the jar complaining at runtime that it's unable to find the libs in path, even after PATH contains the lib files?

Dannyzen, 2024-01-24:
Resolved.

Things I did wrong: I ran `apt-get update` and `apt-get upgrade` after install. This (oddly) broke the installation environment expectations and caused the above issues.

closed 2024-01-24 by Dannyzen

Dannyzen, 2024-01-24:
Actually.... This service may have just needed a PATH variable set in the systemd service config. 

https://stackoverflow.com/questions/63377036/systemd-unable-to-find-java explains how to add a PATH variable at startup.