PhotonVision/photonvision #1160: Snapshot UI could be improved

KangarooKoala, 2024-01-15:
**Describe the bug**
When taking snapshots, the take snapshot button and the camera feed are far apart, requiring either repetitive scrolling or zooming out to go between seeing the camera feed (to move the camera into position) and taking the snapshot.

In addition, there is little feedback about what happens when you take a snapshot- Did it succeed? Did it fail because the chessboard wasn't in frame? Was the click simply missed? Since the snapshot count is close to the take snapshot button, it isn't that bad to determine if the snapshot succeeded, but currently there is no indication for when you try to take a snapshot but the chessboard isn't in frame.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a camera to the coprocessor
2. Connect a laptop to the coprocessor
3. Go to the PhotonVision UI (e.g., photonvision.local:5800)
4. Go to the cameras tab
5. Start taking snapshots

**Screenshots / Videos**
None yet, will hopefully remember to add some later.

srimanachanta, 2024-02-01:
Are you referring to taking calibration snapshots or the normal camera stream snapshots? We  know about the calibration button being annoying on certain laptop displays if that is what you are referring to

KangarooKoala, 2024-02-01:
I'm talking about calibration snapshots.
If there's already issues open for the calibration snapshot button placement and feedback for taking snapshots, then this can be closed as a duplicate- I didn't check for other issues when I made this one.

srimanachanta, 2024-02-02:
Solutions are:
1. Move the "Take Snapshot" and number of snapshots chips below the camera stream. Looks something like this
[image]
2. Fix the camera stream to the right middle of the viewport that way it is constantly in frame even while zooming, this way, we can still change sliders and such and still have view of the cameras to validate changes made there.

KangarooKoala, 2024-02-02:
I personally like option 2 more since it lets you keep more than just the calibration stuff in frame while looking at the camera, but I don't care that much- If anyone has strong reasons to go the other way (maybe some trickiness about the scrolling behavior), I won't object.

PR #1294 (2024-03-21): Sticky calibration camera display card
When UI is not in stacked mode, sticky the camera display card to the top of the display for easy view during calibration.

Fixes #1160
closed 2024-03-21 by mcm001