PhotonVision/photonvision #1164: NetworkManager interface setting is empty by default

rzblue, 2024-01-15:
The "NetworkManager interface" dropdown is empty by default, which causes setting any network settings to silently not work. Ideally, this should be pre-selected with the [first, only, default, etc] network interface, and/or the interface should make it clear that this needs to be set for network settings to take effect.

Platform: Orange Pi
Version: v2024.1.2

mcm001, 2024-01-15:
https://github.com/PhotonVision/photonvision/blob/e1f550a7514701149a209639f39f2b45f55a8587/photon-core/src/main/java/org/photonvision/common/configuration/NetworkConfig.java#L55

Isn't that what this LOC does? Or should do? Journalctl logs from a fresh start (delete config files) of Photon would be helpful.

rzblue, 2024-01-16:
I can get some logs tomorrow. I think when I looked at them earlier, it had fallen back to "Wired connection 1" in the nmcli commands, but there *was* an entry for `netplan-eth0` in the dropdown.

gartaud, 2024-03-11:
We've seen that one too. Thanks for raising this issue @rzblue ;-)

https://github.com/PhotonVision/photonvision/issues/1261 (not raised by our team) might be a duplicate.

mcm001, 2024-11-02:
Addressed by https://github.com/PhotonVision/photonvision/pull/1478 ? Reopen if not

closed 2024-11-02 by mcm001