PhotonVision/photonvision #1167: When an Orange Pi 5 Plus is booted with an "photonvision-v2024.2.0-linuxarm64_orangepi5.img" image file, the system goes into an endless loop and PhotonVision never starts-up.

GBW48, 2024-01-16:
### **Problem Description**
I'm trying to install photonvision-v2024.2.0 on an Orange Pi 5 Plus running an Ubuntu 22.04 from an image that has been specialized for the Orange Pi.  Ubuntu is running from a NVMe drive.  I used the latest version of BalenaEtcher 1.18.11 to flash a high speed 32gb SDcard with a photonvision-v2024.2.0-linuxarm64_orangepi5.img image.  When I try to boot with the imaged SDcard, there is some initial configuration activity on the terminal, but after about 60 seconds, the installation process seems to go into a loop.  The terminal screen screen repeatedly shows the following "elapsed_time_in_seconds rk3x-i2c fec80000.i2c: timeout, ipd: 0x80, state: 3".  This message keeps repeating and never terminates.  I am unable to boot into PhotonVision.

### **To Reproduce**
**Building the SDcard with the photonvision-v2024.2.0-linuxarm64_orangepi5.img image**
1. Download photonvision-v2024.2.0-linuxarm64_orangepi5.img.xz from [GitHub PhotonVision/photonvision](https://github.com/PhotonVision/photonvision/releases) 
2. Use BalenaEtcher 1.18.11 (the latest version) to image a high speed, 32gb SDcard.
3. Insert the imaged SDcard into the SDCard reader on the Orange Pi.
4. When the Orange Pi is turned-on the system boots from the SDcard, and not from the NVMe drive which is used to run Ubuntu 22.04.
5. There is some initial configuration activity on the terminal, but after about 60 seconds, the installation process seems to go into a loop. For example, the terminal screen screen repeatedly shows the following message "[54.309914] rk3x-i2c fec80000.i2c: timeout, ipd: 0x80, state: 3". In subsequent messages, the elapsed time increases, but the remainder of the message remains the same.  These messages keep repeating. I am unable to boot into PhotonVision.

### **Screenshots / Videos**
**Screenshot showing message repetition**
[image]

### **Platform:**
 - Hardware Platform:  Orange Pi 5 Plus with 256gb of memory, 256gb NVMe drive, SDcard reader, HDMI video connector, normal and high speed USB ports, RJ 45 network connector.
 - Software Platform: Orange Pi is running Ubuntu 22.04 LTS.  This version of Ubuntu (ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5-plus.img) was specially built for Orange Pi computers [Ubuntu Rockchip](https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5-plus.html).
 - Network Configuration: Internet assess is provided by a local ethernet connection thru a network switch and cable router.
 - PhotonVision Version: v2024.20. based on photonvision-v2024.2.0-linuxarm64_orangepi5.img
Refer to [GitHub PhotonVision/photonvision](https://github.com/PhotonVision/photonvision/releases).
 - Browser: Chromium version 114.0.7535.35 (Developer Build) Ubuntu 22.04 (64-bit).
 - Camera(s) Used: Arducam 120fps Global Shutter Camera.  For this test, the camera is not connected to the Orange Pi.

### **Additional Comments**
- In response the the following comment [PhotonVision Flashing the Pi Imaghe](https://docs.photonvision.org/en/latest/docs/installation/sw_install/orange-pi.html) "Warning Using an older version of Balena Etcher may cause bootlooping (the system will repeatedly boot and restart) when imaging your Orange Pi. Updating to the latest Balena Etcher will fix this issue", I am using the latest version of Balena Etcher version 1.18.11.
- For my testing I used high speed Samsung 32gb EVO SD cards
- I repeated the building of the photonvision-v2024.2.0-linuxarm64_orangepi5.img image five times.  In each case I used different Samsung 32gb EVO SD card.
- I know the SDcard reader works, because I used it to build an SDcard that runs Ubuntu 22.04.  I used Balena Etcher to image an SDcard from the ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5-plus.img image file.

mcm001, 2024-01-16:
I believe our image is built ONLY for Orange pi 5, and NOT 5+. They're pretty different SBCs. Can you add more warnings to our docs about this?

mcm001, 2024-01-16:
Yep we use this base image: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.30/ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5.img.xz

If you want 5+ support, PR it to photon-image-modifier and then the main repo the same way LL3 support was just added

GBW48, 2024-01-16:
mcm001, thanks for getting back to me.  

Your response indicated, "If you want 5+ support, PR it to photon-image-modifier and then the main repo the same way LL3 support was just added."  Since I'm relatively new to all of this
- What does "PR it to photon-image-modifier and then the main repo" mean?
- Please provide instructions on how to do this.
- Will this issue be fixed, and how will we be notified of the resolution?

Please respond as soon as possible.  I'm working with some students tonight and want to get beyond this issue.
Thanks

GBW48, 2024-01-17:
I did some research on how a Pull Request works. I'm not sure I have the skill and understanding of your build process to produce an image of PhotonVision 2024 that will install properly and run on an Orange Pi 5 Plus coprocessor.

If you could give me the instructions and support (if needed), I'm certainly willing to try to build an image.   Otherwise, could you or one of your collaborators produce an image of PhotonVision 2024 that will run on the Orange 5 Plus system?  There is some urgency here: the FRC team I mentor wants to use PhotonVision for AprilTag detection for this year's FRC competition.  Our first tournament is in less than six weeks!   

Please let me know if what I am requesting is possible and, If so, when we might expect assistance or a resolution to our problem.

I appreciate your help with this.

mcm001, 2024-01-17:
So every other FRC team also has a competition during competition season. We're a volunteer project so if someone has bandwidth and wants to try updating our built to create an image that might happen. 
And as volunteers with an actual life, we don't really like to commit to timelines or feature demands.

In the meantime, you can use this link for instructions for the orange pi 4 but use the 5+ image instead? https://docs.photonvision.org/en/latest/docs/installation/sw_install/orange-pi.html

gerth2, 2024-01-17:
@GBW48 I believe this issue is miscategorized. It was marked as `bug`, but in reality this is a `new-feature` - The 5 Plus is not on the [recommended coprocessors page](https://docs.photonvision.org/en/latest/docs/hardware/selecting-hardware.html#coprocessor-reccomendations). 

While I agree it would be nice to have support, I can't say adding this support is priority above fixing bugs. 

The path forward I recommend is to talk to your team about the tradeoffs- One option is to invest the time in finding a way to get the coprocessor you have on hand working (knowing this is forging a new path that, to my knowledge, no one else has reported success with yet). Alternatively, you can choose a different coprocessor with support.

BytingBulldogs3539, 2024-01-18:
@GBW48 I have created an image file and we will update photonvision in the next update if you can verify for me that this image boots photonvision. https://github.com/PhotonVision/photon-image-modifier/actions/runs/7564887336/artifacts/1177073866

GBW48, 2024-01-18:
@BytingBulldogs3539 Thank you for creating an Orange Pi 5 Plus image file for PhotonVision.  We will test it over the next few days and let you know the results.

JustJoshingYa, 2024-01-18:
If that does not work, you could try Anand's solution detailed in this google doc https://docs.google.com/document/d/1DAPOnU2NfOp91UnMQnkhyiUAanlCDJ6zl_YsCkCMZdA/edit?usp=sharing. 
I used this with my team's orange pi 5+, which worked wonderfully with Apriltag with the same camera you have. The only thing I changed was the ubuntu image https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5-plus.html. I do think the compiled image

GBW48, 2024-01-19:
@JustJoshingYa Thank you for your suggestions. I appreciate it.  Thus far, the compiled image from @BytingBulldogs3539 BytingBulldogs3539 is working for us.

GBW48, 2024-01-19:
@BytingBulldogs3539 Thursday's testing went well.  

We flashed the photonvision_opi5plus.img to an NVMe drive without incident.  Once the NVMe drive was installed, we booted the Orange Pi 5 Plus.  No problems were observed during the boot process.  After logging in and starting the photonvision service, we ran the PhotonVision dashboard from a web browser on a computer connected to the same network.  Then, we set up and calibrated two cameras: one for AprilTag and the other for Colored Shape detection.  Lastly, the students experimented with using the Colored Shape and AprilTag pipelines. 

We will do additional testing on Saturday and let you know the results.

BytingBulldogs3539, 2024-01-20:
> @BytingBulldogs3539 Thursday's testing went well.
>
> We flashed the photonvision_opi5plus.img to an NVMe drive without incident. Once the NVMe drive was installed, we booted the Orange Pi 5 Plus. No [...]

I'm not sure what you mean by logging in and starting the photonvision service that should happen automatically...

closed 2024-01-21 by mcm001

GBW48, 2024-01-21:
@BytingBulldogs3539 
You are correct. When the Orange Pi 5+ is powered on, you don't need to log in and start the photonvision service.  Initially, we thought we had to do this, but today, we realized this was unnecessary.

The students worked with the Photon Vision image (photonvision_opi5plus.img.xz) you created.  **We had no problems with the imaged Orange Pi 5+ during six hours of development and testing.**  Today, the students could read an AprilTag and use the tag's position and orientation to drive the robot toward the observed tag.   Your image just worked!

Thank you for developing a 2024 version of PhotonVision that works with the Orange Pi 5+.  Your efforts are most appreciated!

PR #1170 (2024-01-21): Add Orange Pi 5 Plus
Fixes: #1167