PhotonVision/photonvision #1168: Camera Calibration & Video Mode

WispySparks, 2024-01-17:
**Describe the bug**
When conducting camera calibration there is no indication as to what video mode is used (YUYV vs MJPEG) or a way to choose which video mode to calibrate in. This makes you unable to know which one should be used with your 3D measurements because you don't know which mode the camera was calibrated in.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Cameras'
2. Scroll down to 'Configure New Calibration'
3. Click on 'Resolution'
4. See error

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.
[image]
As seen in the image you can only select the resolution without specifying video mode.

**Platform:**
 - Hardware Platform: Raspberry Pi 4b
 - Network Configuration: Computer connected to RPi through a D-Link radio using ethernet.
 - PhotonVision Version: v2024.2.0
 - Browser: Chrome v120.0.6099.225
 - Camera(s) Used: ELP USBFHD01M Camera (not fisheye), Either -L36 or -L21

mcm001, 2024-01-17:
Ideally the camera video you get back shouldn't actually change between formats. Does the image stretch or change between mjpeg and yuyv? We make the assumption that all resolutions for a camera with the same width and height are valid for a given calibration pretty hard in the background

WispySparks, 2024-01-17:
On the main resolution the numbers I was getting back were different between formats. On another resolution I tested I could visibly see the image stretch/change between formats.

mcm001, 2024-01-17:
Then that's larger in scope than I had initially feared, I think. It sounds like this particular camera actually changes binning between video formats with the same resolution, is that correct?

WispySparks, 2024-01-17:
What is binning?

mcm001, 2024-01-17:
https://en.wikipedia.org/wiki/Pixel_binning

mcm001, 2024-01-17:
So just to confirm. If you leave the camera stationary and keep the same, the same, changing between YUYV and MPEG video modes with the same resolution and everything changes the field of view of the camera? I don't see of an easy way to take account for this in our backend like I mentioned earlier, I think that the easiest solution is going to be to expose a video format drop-down in the UI.

WispySparks, 2024-01-17:
Yes seems like something is wack with my camera..
640x480, YUYV
[image]
640x480, MJPEG
[image]
It's even more noticeable on the 800x600 resolution.

mcm001, 2024-01-17:
Well, that's a new one! Like I mentioned before, this breaks the assumption that we hold in the backend right now that equivalent resolutions have calibrations that can be interchanged. If you have some time and want a fun experiment, you should go poke through that code and see how we could adapt it for your camera.

samfreund, 2025-04-12:
We saw something similar in #505

samfreund, 2025-04-12:
Can we add this as a quirk and then close the issue?

WispySparks, 2025-04-12:
Our solution was to just stop using these weird cameras, but yeah it would make sense to close this issue if a quirk was added

mcm001, 2025-04-12:
How do we think we should handle this via quirks? I suspect that we make some pretty strong assumptions that the same resolution will always work with a given calibration in the backend, so you'd almost have to filter out all MJPG or all YUYV modes.

samfreund, 2025-04-12:
> How do we think we should handle this via quirks? I suspect that we make some pretty strong assumptions that the same resolution will always work with a given calibration in the backend, so you'd almost have to filter out all MJPG or all YUYV modes. 

I think filtering out one of the modes would be a good solution here.