PhotonVision/photonvision #1182: UI partially crashes every few minutes

TheTripleV, 2024-01-21:
**Describe the bug**
The UI stops working after a few minutes of using object detection. The camera feed disappears. The camera selector only shows place holder camera. Refreshing the page does not fix it. Navigating to settings and pressing restart photonvision does work and fix it.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure pipeline to object detection
2. Use it
3. Play around with confidence / brightness / exposure sliders

**Platform:**
 - Orange Pi 5
 - Network Configuration: Connected to frc radio
 - PhotonVision Version: 57f02f31a5b3b652eec7b3de03c2f4d58cb30d27 (the build from the latest main commit)
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Multiple macos machines on latest chrome
 - Camera(s) Used: Idk some USB camera. Not lifecam

mcm001, 2024-01-21:
Cameras disappearing is something I know can happen if you swap between dashboard and any other tab more than 4 or 5 times, and is fixed by a refresh. But this seems different. This is specific to object detection? Are there any specific sliders I can move to guarantee it ought to die? And send journalctl logs from right after a crash and reboot?

TheTripleV, 2024-01-21:
Sorry, I think I reflashed after this issue and so don't have logs at the moment.

We've only been testing object detection. Idk if other mode are affected. I don't think a specific slider made it break. The 3 sliders we were playing with were confidence, exposure, and brightness.

srimanachanta, 2024-01-25:
Usually when the UI freezes up like this, it's separate from what Matt is referring to and is more an uncaught exception in the backend. It's interesting that the restart button then worked though, given the service likely crashed. Without logs of some kind, it's going to be impossible to trace this without reproducing

mcm001, 2024-03-11:
Do dmesg logs from when the crash happens show anything interesting?

cole-wilson, 2024-03-11:
Our team is also having this problem, where the PhotonVision camera stream just shows a spinning circle and NT data drops. I've discussed it [here](https://www.chiefdelphi.com/t/photonvision-losing-connection-orange-pi/458277/2) as well.

closed 2025-07-09 by mcm001