PhotonVision/photonvision #1187: Camera streams don

crschardt, 2024-01-21:
**Describe the bug**
A clear and concise description of what the bug is and what the expected behavior should have been.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64):
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch):
 - PhotonVision Version:
 - Browser (with Version) (Chrome, Edge, Firefox, etc.):
 - Camera(s) Used:

**Additional context**
Add any other context about the problem here.

mcm001, 2024-01-21:
Why installing v4l-utils fixes things is beyond befuddling to me. Would be good to get someone else to reproduce or find the root cause, but it's easy to install for us yeah

crschardt, 2024-01-21:
I'm starting to question that conclusion. I'm doing more testing.

Is it possible that there the streams sometimes don't get released properly when photonvision shuts down?

mcm001, 2024-01-21:
I think possible. Perhaps a longer shutdown delay in systemd is needed?

crschardt, 2024-01-21:
It's turning out to be very intermittent. I can sometimes get it to happen a couple times in a row, but then I'm not able to reproduce it. I'll keep monitoring it to see if I can find some consistent way to reproduce.

crschardt, 2024-08-25:
This hasn't come up recently, so closing.

closed 2024-08-25 by crschardt