PhotonVision/photonvision #1194: PhotonVision image does not correctly identify the network connection name on first boot

crschardt, 2024-01-23:
**Describe the bug**
When booting from a fresh image on the OrangePi5, the photonvision settings page does not show the current network interface in the Network Manager Interface, but it is listed in the dropdown. The user must select it once and after that it is properly saved in settings.

**Screenshots / Videos**
Digging through the logs, on the very first boot, the check for network interfaces returns connection with empty `connName` values.
```
Nov 17 06:45:11 ubuntu java[806]: [2023-11-17 06:45:11] [General - ShellExec] [DEBUG] Executing "nmcli -t -f GENERAL.CONNECTION,GENERAL.DEVICE,GENERAL.TYPE device show"
Nov 17 06:45:11 ubuntu java[806]: [2023-11-17 06:45:11] [General - ShellExec] [DEBUG] Got exit code 0
Nov 17 06:45:11 ubuntu java[806]: [2023-11-17 06:45:11] [General - NetworkUtils] [DEBUG] Found network interfaces:
Nov 17 06:45:11 ubuntu java[806]: [NMDeviceInfo [connName=, devName=eth0, nmType=NMTYPE_ETHERNET], NMDeviceInfo [connName=, devName=lo, nmType=NMTYPE_UNKNOWN]]
```

This causes the code to default to the name "Wired Connection 1" which is used by RaspberryPi. A later event that causes it to scan the network interfaces again yields interfaces with valid `connName` values:
```
Jan 23 03:03:42 photonvision java[806]: [2024-01-23 03:03:42] [General - ShellExec] [DEBUG] Executing "nmcli -t -f GENERAL.CONNECTION,GENERAL.DEVICE,GENERAL.TYPE device show"
Jan 23 03:03:42 photonvision java[806]: [2024-01-23 03:03:42] [General - ShellExec] [DEBUG] Got exit code 0
Jan 23 03:03:42 photonvision java[806]: [2024-01-23 03:03:42] [General - NetworkUtils] [DEBUG] Found network interfaces:
Jan 23 03:03:42 photonvision java[806]: [NMDeviceInfo [connName=netplan-eth0, devName=eth0, nmType=NMTYPE_ETHERNET], NMDeviceInfo [connName=, devName=lo, nmType=NMTYPE_UNKNOWN]]
```

Note the time change. The OPi was connected to my home router and must have synchronized the clock between the two events. 

So, it seems like the first check for network interfaces happens too early and `nmcli` doesn't return valid names, the second check is valid, but by then, PhotonVision has already selected a default value, which is wrong for the OPi.

Options to fix this:
- Special case the default network connection name for OPi
- Update the default network interface when nmcli call returns valid names and the current interface isn't in the list.
- Put a message in the UI asking users to select the correct network interface from the dropdown if it is empty.

mcm001, 2024-01-23:
Duplicate of https://github.com/PhotonVision/photonvision/issues/1164 but this time we have real actual logs, which is super helpful

mcm001, 2024-01-23:
I think that maybe we shouldn't try to guess at a interface name? I'm curious to see how bad the code for (if we see no wired connections at boot, and the user has never ever set the interface, and we see one now; reset the default) is, but I agree that I don't super love the extra backend complexity and state tracking that would require. Idk

mcm001, 2024-11-02:
Addressed by https://github.com/PhotonVision/photonvision/pull/1478 ? Reopen if not

closed 2024-11-02 by mcm001