PhotonVision/photonvision #1198: Bandwidth limit on FRC Radio makes calibration difficult

judsonjames, 2024-01-26:
**Describe the bug**
I've ran into issues with calibrating cameras while connected to the robot's radio. When the robot is configured to use the 4mbs bandwidth limit, every dashboard page load takes several seconds (if not minutes) to load. When calibrating a new camera, I'm stuck loading the "Placeholder Camera" waiting for the websocket to finalize. While the camera is loading for calibration, or when I load the dashboard, I'm seeing spikes well above 4mbs in the Performance Monitor of Windows. I have replicated this both on a physical robot with all other components, and a test board using the minimum components needed (PDB, RIO, Radio, Switch, Coprocessor). On the test board, I'm powering the coprocessor and switch independently of the battery, removing power supply issues out.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [FRC Radio Configuration](https://docs.wpilib.org/en/stable/docs/zero-to-robot/step-3/radio-programming.html)
2. Follow instructions on flashing the Radio, ensure to keep the bandwidth limit enabled. I tested using the 5ghz mode.
3. Plug Radio port 1 into network switch, RIO and coprocessors into other network switch ports.
4. Connect to radio from Driver Station via WiFi
5. Open the Performance Monitor in Windows (or use WireShark) to monitor the total bytes/sec. WPILib has documentation on how to do this [here](https://docs.wpilib.org/en/stable/docs/networking/networking-introduction/measuring-bandwidth-usage.html)
6. Let Performance Monitor run for a minute or two to get a baseline. This should be in the 1mbs range if you're running just the radio, RIO, and coprocessor on the same switch.
7. Open the PhotonVision dashboard in browser (I used Chrome)
8. Watch Performance Monitor cap at 4mbs for several seconds (if not minutes) until the backend connects.
9. Switch cameras and see it continue to spike.
10. Open Calibration menu, and reload the web page
11. Camera backend socket may not connect.
12. Replicate steps 2 through 11, but turn off the 4mbs bandwidth limit. The problem is virtually gone.

**Screenshots / Videos**
Loading page, then drops when cameras are connected, stays above 4mbs when in calibration mode.
[image]

Repeated refreshes of the web dashboard trying to get cameras to load on 4mbs bandwidth limit
[image]

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): OrangePi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Radio -> Network Switch <- everything else. Laptop with driver station on WiFi.
 - PhotonVision Version: 2024.2.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Arducam OV9281 (x2), OV9782 (x1)

**Additional context**
At home, teams should be able to disable their 4mbs limit at home to get around this, but if they need to recalibrate cameras at competitions, they may need to impose the 4mbs limit, adding time needed to calibrate.

I don't think this is perse an issue with PhotonVision, but an issue with the 4mbs limitation in FRC. It may be relatively easy to fix on PV's end to make it easier for teams to calibrate multiple cameras.

gerth2, 2024-01-26:
Agree with the overall intent - that's a lot of data getting transferred that shouldn't cause UI hangs, that needs reworked in general. 

> At home, teams should be able to disable their 4mbs limit at home to get around this, but if they need to recalibrate cameras at competitions, they may need to impose the 4mbs limit, adding time needed to calibrate.

Does connecting over ethernet sidestep this? It may be a better workaround for this season at least.

judsonjames, 2024-01-26:
> Does connecting over ethernet sidestep this? It may be a better workaround for this season at least.

Yes, connecting my laptop directly to the switch sidesteps the issue.

mcm001, 2024-01-26:
I think I have got some ideas to (for now) reduce the data we send quite a bit. My cal refactor to collect all of the data is most likely responsible here lol. I'll see what low hanging fruit I can find

mcm001, 2024-01-26:
[image]
so with 60 targets in view, every time we update the UI sends 15.3 KB of data. most of it seems empty. i bet protobuf or Packet would make this -so- much smaller

```
<=Sending: {updatePipelineResult={0={latency=14.365, fps=28.0, classNames=[], targets=[{area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, a [... cut ...] 0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}, {area=0.0, ambiguity=-1.0, classId=-1, pose={angle_z=0.0, angle_y=0.0, angle_x=0.0, qw=1.0, qx=0.0, x=0.0, qy=0.0, y=0.0, qz=0.0, z=0.0}, confidence=-1.0, fiducialId=-1, pitch=0.0, yaw=0.0, skew=0.0}]}}}
```
like 99 percent of this is useless data we dont actually need in the frontend

mcm001, 2024-01-26:
And on sending full settings with 36 images:
```
Sending: {settings={networkSettings={networkingDisabled=false, staticIp=, runNTServer=false, shouldManage=true, ntServerAddress=0, connectionType=0, networkInterfaceNames=[NMDeviceInfo [connName=Wired connection 3, devName=enxa0cec8dc851c, nmType=NMTYPE_ETHERNET]], canManage=true, hostname=photonvision, networkManagerIface=Wired connection 1, setDHCPcommand=nmcli con mod ${interface} ipv4.method "auto" ipv6.method "disabled", shouldPublishProto=false, setStaticCommand=nmcli con mod ${interface} ipv4.addresses ${ipaddr}/8 ipv4.method "manual" ipv6.method "disabled"}, general={gpuAcceleration=, hardwarePlatform=Linux x64, rknnSupported=false, hardwareModel=, mrCalWorking=true, version=v2024.2.2}, atfl=edu.wpi.first.apriltag.AprilTagFieldLayout@cfe44485, lighting={brightness=100, supported=false}}, cameraSettings=[{inputStreamPort=1181, currentPipelineIndex=0, outputStreamPort=1182, cameraQuirks=QuirkyCamera [baseName=HP Wide Vision HD Camera, displayName=, usbVid=1480, usbPid=991, quirks={AWBGain=false, ArduOV2311=false, AdjustableFocus=false, ArduCamCamera=false, PiCam=false, ArduOV9281=false, StickyFPS=false, Gain=false, FPSCap100=false, CompletelyBroken=false}], isFovConfigurable=true, fov=70.0, currentPipelineSettings={offsetRobotOffsetMode=0, streamingFrameDivisor=0, offsetDualPointBArea=0.0, contourGroupingMode=0, contourFilterRangeY=2.0, contourFilterRangeX=2.0, hsvValue=org.photonvision.common.util.numbers.IntegerCouple@471d2a79, cameraGain=-1, cameraBlueGain=-1, cameraRedGain=-1, cornerDetectionSideCount=4, contourRatio=org.photonvision.common.util.numbers.DoubleCouple@53a1cd98, contourTargetOffsetPointEdge=0, pipelineNickname=New Pipeline, inputImageRotationMode=0, contourArea=org.photonvision.common.util.numbers.DoubleCouple@987bc79, solvePNPEnabled=false, contourFullness=org.photonvision.common.util.numbers.DoubleCouple@1f6ea946, pipelineIndex=0, inputShouldShow=false, cameraAutoExposure=false, contourSpecklePercentage=5, contourTargetOrientation=1, targetModel=2, cornerDetectionUseConvexHulls=true, outputShouldShow=true, outputShouldDraw=true, offsetDualPointA={0.0, 0.0}, offsetDualPointB={0.0, 0.0}, hsvHue=org.photonvision.common.util.numbers.IntegerCouple@318e1630, ledMode=true, hueInverted=false, outputShowMultipleTargets=false, contourSortMode=0, cameraExposure=6.0, offsetSinglePoint={0.0, 0.0}, cameraBrightness=50, offsetDualPointAArea=0.0, cornerDetectionExactSideCount=false, cameraVideoModeIndex=0, cornerDetectionStrategy=0, cornerDetectionAccuracyPercentage=10.0, hsvSaturation=org.photonvision.common.util.numbers.IntegerCouple@6c9ed094, pipelineType=2, contourIntersection=1}, isCSICamera=false, calibrations=[CameraCalibrationCoefficients [resolution=1280x720, cameraIntrinsics=JsonMat [rows=3, cols=3, type=6, data=[773.8217310577863, 0.0, 636.9463295609688, 0.0, 773.8705151375215, 372.638834457646, 0.0, 0.0, 1.0], wrappedMat=Mat [ 3*3*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7faa900b6cb0, dataAddr=0x7faa90053f80 ], wpilibMat=null, wrappedMatOfDouble=null], distCoeffs=JsonMat [rows=1, cols=8, type=6, data=[0.016926942898196885, -0.09179894736714628, 0.0012960292020820594, -0.002412144550308628, 0.07563035539923747, -0.00826761554377048, 0.002512437408769198, -0.0016716668181851603], wrappedMat=Mat [ 1*8*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7faa900b6af0, dataAddr=0x7faa90027080 ], wpilibMat=null, wrappedMatOfDouble=Mat [ 1*8*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7faa900b6a80, dataAddr=0x7faa900b6840 ]], observations=[org.photonvision.vision.calibration.BoardObservation@6d607dd4, org.photonvision.vision.calibration.BoardObservation@55cf9926, org.photonvision.vision.calibration.BoardObservation@4ffdfea0, org.photonvision.vision.calibration.BoardObservation@768563bf, org.photonvision.vision.calibration.BoardObservation@18653e66, org.photonvision.vision.calibration.BoardObservation@31ca92e5, org.photonvision.vision.calibration.BoardObservation@11627282, org.photonvision.vision.calibration.BoardObservation@4ca7738a, org.photonvision.vision.calibration.BoardObservation@7d00e276, org.photonvision.vision.calibration.BoardObservation@18b80a89, org.photonvision.vision.calibration.BoardObservation@1a4c8d7e, org.photonvision.vision.calibration.BoardObservation@6d00869e, org.photonvision.vision.calibration.BoardObservation@1769da1a, org.photonvision.vision.calibration.BoardObservation@4fdd663c, org.photonvision.vision.calibration.BoardObservation@310425c4, org.photonvision.vision.calibration.BoardObservation@58d9610c, org.photonvision.vision.calibration.BoardObservation@1b9c907, org.photonvision.vision.calibration.BoardObservation@2571a38a, org.photonvision.vision.calibration.BoardObservation@354b88e5, org.photonvision.vision.calibration.BoardObservation@26895e41, org.photonvision.vision.calibration.BoardObservation@69b75a66, org.photonvision.vision.calibration.BoardObservation@4172d929, org.photonvision.vision.calibration.BoardObservation@28177d42, org.photonvision.vision.calibration.BoardObservation@44005557, org.photonvision.vision.calibration.BoardObservation@173753db, org.photonvision.vision.calibration.BoardObservation@405b7b85, org.photonvision.vision.calibration.BoardObservation@5335f3a1, org.photonvision.vision.calibration.BoardObservation@15d9be7d, org.photonvision.vision.calibration.BoardObservation@38dd9e96, org.photonvision.vision.calibration.BoardObservation@23415d5b, org.photonvision.vision.cali [... cut ...] width=320, fps=30, height=240}, 4={pixelFormat=YUYV, width=352, fps=30, height=288}, 5={pixelFormat=MJPEG, width=352, fps=30, height=288}, 6={pixelFormat=YUYV, width=640, fps=30, height=360}, 7={pixelFormat=MJPEG, width=640, fps=30, height=360}, 8={pixelFormat=YUYV, width=640, fps=30, height=480}, 9={pixelFormat=MJPEG, width=640, fps=30, height=480}, 10={pixelFormat=MJPEG, width=1280, fps=30, height=720}}, pipelineNicknames=[New Pipeline], nickname=HP_Wide_Vision_HD_Camera}]}
Bytes: 12099814
```

gerth2, 2024-02-02:
Per testing on my end this afternoon.... really does seem like this got fixed on 62112cd2fd8e03ca144c47b985d23b7f67304863 - ok to close out?

mcm001, 2024-02-02:
Closed by #1200

closed 2024-02-02 by mcm001