PhotonVision/photonvision #1199: Calibrating to a resolution different to the one selected

SuperSaurus1, 2024-01-26:
**Describe the bug**
When a resolution of a higher resolution, ex. 1280x720 is selected for calibration, upon completion a calibration dataset is created for 424x240. Within the logs it sets the resolution for calibration to the lower resolution, despite the UI displaying the desired resolution.

**To Reproduce**
Steps to reproduce the behavior:
1. Run calibration
2. Look at the data set
3. Look at the logs

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.
[image]
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/14069844/photonvision-journalctl.txt)

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5 x 4
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Network Switch
 - PhotonVision Version: 2024.2.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Spinel UC10MPC x 4

**Additional context**
This was replicated on both a clean image of the photon vision image, then with debian installation, with the exact same output, on all 4 of the Orange Pi's plugged in.

mcm001, 2024-01-27:
I think i need to see whats going on inside the UI too, Might need to set up a local dev environment and set some breakpoints inside the UI or add console.log's and stuff to troubleshoot this one

SuperSaurus1, 2024-01-31:
Not going to lie, not super sure what you mean by that but I'd be happy to help however I can

mcm001, 2024-01-31:
Better question to start out with since it's easier. Does this happen on a windows or Linux laptop or desktop, too?

The next step would be finding the spot in the frontend where the request to enter calibration is sent (probably inside of CameraCalibrationCard) and adding print statements to see if it does what you expect. If you put these changes in a pull request, GitHub actions will build a JAR that you can upload using offline update to test out, and you can look at the browser console to see your prints. That's how I would approach it

samfreund, 2025-04-14:
@mcm001 close as stale? AFAIK this isn't a thing anymore.

closed 2025-04-14 by Gold856