PhotonVision/photonvision #1202: Unable to flash Orange Pi image to SD Card with balenaEtcher

agasser, 2024-01-27:
**Describe the bug**
Unable to flash Orange Pi image to SD Card with balenaEtcher.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [photonvision-v2024.2.2-linuxarm64_orangepi5.img.xz](https://github.com/PhotonVision/photonvision/releases/download/v2024.2.2/photonvision-v2024.2.2-linuxarm64_orangepi5.img.xz)
2. Try to flash the image to a micro SD Card with balenaEtcher 1.18.11

balenaEtcher goes through the writing step, but it fails toward the end of the verify step with the error below.

**Screenshots / Videos**
[image]

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Windows 11 x64
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): NA
 - PhotonVision Version: v2024.2.2 (also happens with v2024.2.1)
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: NA

agasser, 2024-01-27:
I gave it a try with the Raspberry Pi Imager, and it was successful. I don't know what balenaEtcher didn't like about the image. I even tried reinstalling balenaEtcher.

mcm001, 2024-01-27:
weird. lemme try balenaetcher on linux, version 1.18.11

mcm001, 2024-01-27:
Cant reproduce on linux - closing for now

closed 2024-01-27 by mcm001