PhotonVision/photonvision #1203: "Reset Samples" button for standard deviations freezes the page

Gold872, 2024-01-28:
**Describe the bug**
On the web interface, if multiple tags are in view, the standard deviation always stays at "NaN/100 Samples", and hitting the "Reset Samples" button freezes the page, I have to refresh it to be able to change any other settings or view camera streams.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up an 3D April Tag pipeline with multi-tag estimation turned on
2. Have multiple tags in view
3. Click on "Reset Samples"

**Screenshots / Videos**
[image]

https://github.com/PhotonVision/photonvision/assets/91761103/2ba8252a-22d5-434c-9df3-d8955121b0fb

**Platform:**
 - Hardware Platform: Windows 10 Home 22H2
 - Network Configuration: Windows Localhost
 - PhotonVision Version: 2024.2.2
 - Browser: Chrome 121.0.6167.85
 - Camera Used: Microsoft Lifecam HD 3000

**Additional context**
Add any other context about the problem here.

This also seems to happen when running on a Raspberry Pi, but I couldn't get any footage of it.

DevonRD, 2024-03-22:
This problem appears to happen regardless of whether there are targets in view, and seems to me like it is an infinite loop problem since both the backend and frontend throw no errors. Investigating further

DevonRD, 2024-03-22:
Hunch was right, but that was a symptom of the real issue and not the cause. Fix incoming

PR #1299 (2024-03-23): [Client] Fix issue with clearing multitag buffer
The state multitagResultBuffer dictionary uses integer keys, and stringifying the reference key yielded null results. This caused an infinite while loop on "Reset Samples" (null is always != 0) and caused the entire multitag pose standard deviation section to display the nullish-coalesced default values. Fixed the reference and added in a nullish coalescing condition on the suspect while loop to avoid the frontend lockup if this is somehow encountered again in the future

Fixes #1203
closed 2024-03-23 by srimanachanta