PhotonVision/photonvision #1204: Support for OV9782

TheTripleV, 2024-01-28:
OV9782 is a color global shutter camera.

It's exposure range is 1-60.
It has hue, saturation, gamma, white balance, and contrast control.

The default white balance is pretty bad. 3500 is a better default.
The camera also supports auto white balance which seems to work well and changes near instantly.
It would be nice to have an updated white balance default, white balance slider, and a checkbox for auto whitebalance.

The camera has a reddish tinge (most noticeable around white lights and gray backgrounds). It never fully goes away but correcting white balance mostly resolves it

Potential performance improvement for apriltags: setting saturation to 0 on the sensor does spit out a grayscale image. If this is happening on the sensor itself, it would save on an op for apriltag mode.

output of lsusb for vid pid: `ID 0c45:6366 Microdia Webcam Vitade AF`

output of `v4l2-ctl --all`:
```
Driver Info:
	Driver name      : uvcvideo
	Card type        : Arducam OV9782 USB Camera: Ardu
	Bus info         : usb-fc800000.usb-1
	Driver version   : 5.10.160
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : Arducam OV9782 USB Camera: Ardu
	Serial           : UC852
	Bus info         : usb-fc800000.usb-1
	Media version    : 5.10.160
	Hardware revision: 0x00000100 (256)
	Driver version   : 5.10.160
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Arducam OV9782 USB Camera: Ardu
	Function         : V4L2 I/O
	Flags            : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000013: from remote pad 0x100000a of entity 'Extension 4' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 320/240
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 154189
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 320, Height 240
	Default     : Left 0, Top 0, Width 320, Height 240
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 320, Height 240, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 320, Height 240, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=90 value=90
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=0
                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4000
                      sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=3 value=3
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1 (Manual Mode)
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=157 value=1000
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
```

mcm001, 2024-01-28:
So exposure_absolute is actually 0-30 not 0-5000?

TheTripleV, 2024-01-28:
0-60. The max is in the 55-60 range but I can't tell the difference between those top few.

mcm001, 2024-01-30:
gotcha, whoops typo. Anyways, great to confirm that it enumerates the same as other arducam cameras so we will need to do the same cameras tab hack as with the 2311/9281. should be super straight forwards to:

add a quirky camera
[image]
[image]

add it to the UI dropdown
[image]
[image]

and add the implementation with a check for if a camera has the OV9782 quirk to UsbCameraSource.java

PR #1284 (2024-07-01): Adds support for OV9782's quirks
The OV9782 camera has a specific exposure range, so a camera quirk for it needs to exist. The default white balance is also pretty bad, so it must be adjusted.

Closes #1204
commit 173b6d9c (2024-07-01): Adds support for OV9782's quirks (#1284)

The OV9782 camera has a specific exposure range, so a camera quirk for
it needs to exist. The default white balance is also pretty bad, so it
must be adjusted.

Closes #1204

---------
closed 2024-07-01 by Juniormunk