PhotonVision/photonvision #1208: Allow running ML and Apriltags on a single camera

TheTripleV, 2024-01-30:
If a color camera is plugged in, I think both object detection and apriltag detection should be allowed to occur at the same time.

I know almost nothing of the PhotonVision internals so a couple approaches I think are possible:

#### 1. Add a new AprilTagPlusObjectDetectionPipeline
Pros:
- Quickest to implement?

Cons:
- The 2 pipelines are hardcoded (Apriltag, Object). There'll be no option to swap in Aruco, Aruco Nano.
- The 2 pipelines will run sequentially, increasing latency (and probably halving framerate)
Can 2 persistent threads be created to feed the apriltag detector and the NPU in parallel?

#### 2. Clone the video source using v4l2loopback
Pros:
- No change to most of PhotonVision pipelines. The won't know anything's different.
- 2 pipelines can't both control camera settings so the dummy input will need to be marked as virtual and it's inputs have to be grayed out in the UI.

Cons:
- Allowing a loopback device to set parameters and resolution might be hard to do since the v4l2loopback consumers don't have write access to the producer.

#### 3. Make a copy of the cvSink Source
Pros:
- No change to most of PhotonVision pipelines. The won't know anything's different.
- 2 pipelines can't both control camera settings so the dummy input will need to be marked as virtual and it's inputs have to be grayed out in the UI.

Cons:
- cscore has a copySource but it's not documented so idk if it would work here

mcm001, 2024-01-30:
I think copying the cvSink and/or creating a new DuplicateOfAUSBCameraFrameProvider would be the lowest friction way to do this? I dont think ive ever heard of anyone using copySource in cscore before so you should mess with that. Any way you go though youll need a controller/subscriber model where one pipeline can control camera hardware settings and the other isn't allowed to

samfreund, 2025-03-27:
There's an implementation [here], looks like it's Reefscape specific.

mcm001, 2025-03-27:
I also took a stab at this in a branch

https://github.com/BREAD5940/photonvision-fork/commit/de3ede534a7527cabbe1adbf0beb895b6906b311