PhotonVision/photonvision #1210: photonlibpy usage is not type checkable

auscompgeek, 2024-02-02:
**Describe the bug**

It isn't possible to type check code that imports `photonlibpy`.

**To Reproduce**

```sh
echo "from photonlibpy.photonCamera import PhotonCamera" > test.py
python -m mypy test.py
```

**Screenshots / Videos**

```
test.py:1: error: Skipping analyzing "photonlibpy.photonCamera": module is installed, but missing library stubs or py.typed marker  [import-untyped]
test.py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
Found 1 error in 1 file (checked 1 source file)
```

**Platform:**
 - PhotonVision Version: 2024.2.0

**Additional context**

- [PEP 561](https://peps.python.org/pep-0561/)
- https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports

LucienMorey, 2024-11-20:
fyi, I think this is no longer an issue since #1570. It finished off the last of the type checking things and there is now active ci checking it on each pr. At the very least the steps to reproduce now show:

```
echo "from photonlibpy.photonCamera import PhotonCamera" > test.py
python -m mypy test.py
Success: no issues found in 1 source file
```

auscompgeek, 2024-11-20:
I'm still able to reproduce:

```console
> ls
test.py
> cat test.py
from photonlibpy.photonCamera import PhotonCamera
> python -m mypy test.py
test.py:1: error: Skipping analyzing "photonlibpy.photonCamera": module is installed, but missing library stubs or py.typed marker  [import-untyped]
test.py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
Found 1 error in 1 file (checked 1 source file)
> python -m pip list
Package           Version
----------------- ----------
mypy              1.13.0
mypy-extensions   1.0.0
numpy             2.1.3
opencv-python     4.10.0.84
photonlibpy       2025.0.0b4
pip               24.2
pyntcore          2025.0.0b1
robotpy-apriltag  2025.0.0b1
robotpy-cli       2024.0.0
robotpy-cscore    2025.0.0b1
robotpy-hal       2025.0.0b1
robotpy-wpimath   2025.0.0b1
robotpy-wpinet    2025.0.0b1
robotpy-wpiutil   2025.0.0b1
typing_extensions 4.12.2
wpilib            2025.0.0b1
```

LucienMorey, 2024-11-20:
> I'm still able to reproduce:
> 
> ```
> > ls
> test.py
> > cat test.py
> from photonlibpy.photonCamera import PhotonCamera
> > python -m mypy test.py
> test.py:1: error: Skipping analyzing "photonlibpy.photonCamera": module is installed, but missing library stubs or py.typed marker  [import-untyped]
> test.py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
> Found 1 error in 1 file (checked 1 source file)
> > python -m pip list
> Package           Version
> ----------------- ----------
> mypy              1.13.0
> mypy-extensions   1.0.0
> numpy             2.1.3
> opencv-python     4.10.0.84
> photonlibpy       2025.0.0b4
> pip               24.2
> pyntcore          2025.0.0b1
> robotpy-apriltag  2025.0.0b1
> robotpy-cli       2024.0.0
> robotpy-cscore    2025.0.0b1
> robotpy-hal       2025.0.0b1
> robotpy-wpimath   2025.0.0b1
> robotpy-wpinet    2025.0.0b1
> robotpy-wpiutil   2025.0.0b1
> typing_extensions 4.12.2
> wpilib            2025.0.0b1
> ```

Yep I was running everything with the internal build and test scripts but didn't test with the package from pypi... I will need to look at that next

closed 2024-11-21 by mcm001

LucienMorey, 2024-11-21:
I just tried to run this check with `2025.0.0b5` being stamped, and everything was good. @auscompgeek are things good for you as well?

commit fa66ed86 (2024-11-21): add file marker for type checking (#1598)

I think this is the correct way to do this based on my understanding of
the guide to making a [pep561 compliant
package](https://mypy.readthedocs.io/en/stable/installed_packages.html#creating-pep-561-compatible-packages)
and this example in
[black](https://github.com/psf/black/pull/1395/files). Given that I get
all the type-checking info from a local installation, I don't know if
this is testable until it's published on Pypi.

resolves #1210