PhotonVision/photonvision #1224: Can't get camera to detect on Raspberry Pi

wojtase1, 2024-02-07:
I am trying to get the latest version of PhotonVision to run on a Raspberry Pi.  I have managed to get it to run on Windows and MacOS. When running on the Pi I can get the web interface but it does not seem to recognize the camera.  
I have updated the pi with 'apt upgrade' and I downloaded the latest photonvision jar file from GitHub.  I know that the OS recognizes the camera because the I run 'v4l2-ctl --list-devices' I get.

Microsoft® LifeCam HD-3000: Mi (usb-xhci-hcd.1-1):
	/dev/video0
	/dev/video1
	/dev/media3

But on the web guide I just get the yellow box with a spinning wheel and there are no options for cameras.  Just "Placeholder" in the dropdown.  The funny thing is on my Mac it lets me pick everything from the built in camera, life cam, and even my iPhone!
[image]

crschardt, 2024-02-07:
Would you please try it again by flashing the latest image as is? Don’t run apt update.

wojtase1, 2024-02-07:
Yep that did it.  Apparently I went over the top.  Thanks.