PhotonVision/photonvision #1225: Calibration has wrong list of resolutions

KangarooKoala, 2024-02-07:
**Describe the bug**
When calibrating, the list of available resolutions is incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug OV9281 into Orange Pi 5
2. Go to the 'Dashboard' tab
3. Record/Remember available resolutions
4. Go to the 'Cameras' tab
5. Note that available resolutions for new calibration are different

**Screenshots / Videos**
Video showing both the actual resolutions (in the Dashboard tab) and the calibration resolutions (in the Cameras tab):

https://github.com/PhotonVision/photonvision/assets/91924258/4987f69f-b1ee-4430-a34c-7384cc9a86b4

**Platform:**
 - Hardware Platform: Orange Pi 5, Windows
 - Network Configuration: Ethernet from OPi to radio, then ethernet from radio to laptop
 - PhotonVision Version: 2024.2.4
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome 120.0.6099.225 
 - Camera(s) Used: OV9281

**Additional context**
I also had to refresh to even get the calibration UI in the cameras tab, which may be from the same root issue.

mcm001, 2024-02-07:
I've seen this reported twice now. What do you see in the browser console? And any chance you could debug the bit of UsbCameraSource where we enumerate video modes?

I really wish my C920+lifecam setup at home would let me reproduce, but alas. And the ov9281 is totally gone from Amazon

crschardt, 2024-02-07:
A couple questions:

1. Do you have one or two cameras connected?
2. How did you upgrade to 2024.2.4?
3. Were the camera(s) already calibrated?

KangarooKoala, 2024-02-07:
> I've seen this reported twice now. What do you see in the browser console? And any chance you could debug the bit of UsbCameraSource where we enumerate video modes?

Unfortunately, I won't be able to check the browser console until the next meeting, but I'd be glad to help debug this issue! What would the debugging process look like? Sadly, I'm using a school laptop at meetings which has restricted permissions- I tried running `./gradlew photon-server:shadowJar --max-workers 2 -PArchOverride=linuxarm64` (based on [the CI](https://github.com/PhotonVision/photonvision/blob/v2024.2.4/.github/workflows/build.yml#L253)) earlier today, but I got an IOException for the `photon-targetting:generateProto` task since it couldn't run the protoc executable.

> A couple questions:
> 1. Do you have one or two cameras connected?
> 2. How did you upgrade to 2024.2.4?
> 3. Were the camera(s) already calibrated?

1. One camera
2. I flashed the orange pi with the linuxarm64-orangepi5 2024.2.3 image earlier, and then uploaded the linuxarm64 2024.2.4 jar and restarted the photonvision service (`sudo systemctl restart photonvision.service`).
3. I had calibrated the camera earlier on different versions, though I don't remember which ones at the moment.

KangarooKoala, 2024-02-10:
I did some more testing, and it seems to be based on the settings. Here's what I did:
1. Flash with 2024.2.3 orange pi 5 image
2. Update jar to 2024.2.4
3. Export settings
3. Calibration works fine
4. Import [these settings](https://github.com/PhotonVision/photonvision/files/14230153/photonvision-settings-2024-02-06.zip) from previous versions (I don't remember exactly which versions I did stuff on)
5. Calibration resolutions are broken
6. Import settings from step 3 ([photonvision-settings-export.zip](https://github.com/PhotonVision/photonvision/files/14230163/photonvision-settings-export.zip))
7. Calibration works fine

samfreund, 2025-04-08:
We need to confirm that this is still an issue with 2025.3.1, if not, we can close it.

KangarooKoala, 2025-04-08:
Seems to have been fixed.

closed 2025-04-08 by KangarooKoala