PhotonVision/photonvision #1231: 3d bounding box is distorted when top box is behind camera

gerth2, 2024-02-11:
**Describe the bug**
Strange distortion for apriltags that are *very* close

[image]

all good

[image]

Still fine

[image]

oop

**To Reproduce**
Steps to reproduce the behavior:

Calibrate 3d on any camera

Get an apriltag very close to the camera

**Screenshots / Videos**
see above

**Platform:**
 - pi3

**Additional context**
I'm assuming this is because `Calib3d.projectPoints` does not handle the negative X real world points case well? Everything else is a floating point number so it shouldn't be that bad?

mcm001, 2024-02-12:
Yeah, a brief googling indicates that this breaks things. How do we draw the wire frame field for simulation? That has lines out of the camera FOV

mcm001, 2024-02-12:
> In some cases the distortion values you get from calibrateCamera result in a distortion function that is non-monotonic and the image (or projected points) can “wrap around”…So this could be what you are seeing. Maybe.

https://forum.opencv.org/t/projectpoints-point-visibility-with-distortion/7055/3

amquake, 2025-11-04:
> Yeah, a brief googling indicates that this breaks things. How do we draw the wire frame field for simulation? That has lines out of the camera FOV

See https://github.com/PhotonVision/photonvision/blob/main/photon-lib/src/main/java/org/photonvision/simulation/VideoSimUtil.java#L462

Effectively we use the known camera field of view (its 'frustum') to clip any lines going outside of it (related to 'frustum culling'). If we want to draw a 3d line that goes outside of the frustum, we find any point where that line intersects the frustum boundary and clip the line endpoint to there. That way, we only draw the 2d line which is the projection of the 3d line portion visible to the camera.

I know this is very relevant to game engines and graphics so I wonder if there is a simpler approach and/or a way to do this for images as well without straying too far beyond plain opencv. We could use the current simulation implementation for the 3d box visualization as well, but I'm a little concerned about the performance implications.