PhotonVision/photonvision #1233: PV Camera Calibration Broken for See3CAM_CUG

randomstring, 2024-02-12:
**Describe the bug**
A clear and concise description of what the bug is and what the expected behavior should have been.

**To Reproduce**
Steps to reproduce the behavior:
1. use USB See3CAM_CUG camera
2. click on "Start Calibration"
3. UI does not bring up calibration page.

Log output:
```
[2024-02-12 11:36:39] [VisionModule - VisionModule - See3CAM_24CUG_3] [INFO] Starting calibration at resolution index 1 and settings UICalibrationData{videoModeIndex=1, count=0, minCount=12, hasEnough=false, squareSizeIn=1.0, patternWidth=8, patternHeight=8, boardType=CHESSBOARD}
[2024-02-12 11:36:39] [VisionModule - VisionModule - See3CAM_24CUG_3] [INFO] Setting pipeline to -2
[2024-02-12 11:36:39] [VisionModule - VisionModule - See3CAM_24CUG_3] [INFO] Pipeline name: New Pipeline
[2024-02-12 11:36:39] [General - ShellExec] [DEBUG] Executing "nmcli -t -f GENERAL.CONNECTION,GENERAL.DEVICE,GENERAL.TYPE device show"
[2024-02-12 11:36:39] [General - ShellExec] [DEBUG] Got exit code 0
[2024-02-12 11:36:39] [General - NetworkUtils] [DEBUG] Found network interfaces:
[NMDeviceInfo [connName=netplan-eth0, devName=eth0, nmType=NMTYPE_ETHERNET], NMDeviceInfo [connName=, devName=lo, nmType=NMTYPE_UNKNOWN]]
[2024-02-12 11:36:39] [VisionModule - VisionSourceSettables] [INFO] Setting video mode to FPS: 120 Width: 1280 Height: 720 Pixel Format: kMJPEG
[2024-02-12 11:36:39] [VisionModule - VisionSourceSettables] [DEBUG] Setting camera exposure to 20
[2024-02-12 11:36:39] [VisionModule - VisionSourceSettables] [DEBUG] Setting auto exposure to true
[2024-02-12 11:36:39] [VisionModule - VisionModule - See3CAM_24CUG_3] [ERROR] Unable to set camera auto exposure!
[2024-02-12 11:36:39] [VisionModule - VisionModule - See3CAM_24CUG_3] [ERROR] VideoException [edu.wpi.first.cscore.VideoException: property write failed]
```

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

**Platform:**
 - Hardware Platform: Tried this on a RPi5 and an OPi5
 - Network Configuration: Ethernet, static IP address
 - Browser : MAcOS Firefox 122.0, and Chrome
 - Camera(s) Used: See3CAM_CUG
 
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/14249745/photonvision-journalctl.txt)
[photonvision-settings-export (1).zip](https://github.com/PhotonVision/photonvision/files/14249773/photonvision-settings-export.1.zip)

randomstring, 2024-02-12:
I'm seeing this error everytime I load the /#/cameras tab. maybe relevant?

```
GeneralSettingsStore.ts:100 TypeError: Reduce of empty array with no initial value
    at Array.reduce (<anonymous>)
    at CameraCalibrationCard.vue:29:40
    at Array.forEach (<anonymous>)
    at Object.n (CameraCalibrationCard.vue:22:68)
    at o.DW (CameraCalibrationCard.vue:229:25)
    at e._render (vue.runtime.esm.js:2700:28)
    at o.i (vue.runtime.esm.js:3891:27)
    at e.get (vue.runtime.esm.js:3462:33)
    at e.run (vue.runtime.esm.js:3538:30)
    at Fw (vue.runtime.esm.js:4141:17)
```

mcm001, 2024-02-12:
That particular error is probably fixed in https://github.com/PhotonVision/photonvision/pull/1229 (merged, just waiting for CI)

mcm001, 2024-02-13:
did the latest dev fix things?

randomstring, 2024-02-13:
Yes, problem fixed.  I had a student test it last night at our meeting.