PhotonVision/photonvision #1236: NetworkManager doesn't work on Orange Pi 5

KangarooKoala, 2024-02-13:
**Describe the bug**
NetworkManager isn't detecting any wired connections.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the settings tab
2. Scroll down to "NetworkManager interface"
3. Note the banner "Photon cannot detect any wired connections! Please send program logs to the developers for help"
4. Observe that changing the network settings and clicking save doesn't change the settings

**Screenshots / Videos**
Video:

https://github.com/PhotonVision/photonvision/assets/91924258/d61a20db-3e9d-450f-b9f2-47d9bff61304

Settings:

[photonvision-settings-export.zip](https://github.com/PhotonVision/photonvision/files/14254947/photonvision-settings-export.zip)

**Platform:**
 - Hardware Platform: Orange Pi 5, Windows
 - Network Configuration: Ethernet from OPi to radio, then ethernet from radio to laptop
 - PhotonVision Version: 2024.2.4
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome 120.0.6099.225
 - Camera(s) Used: Arducam OV9281

**Additional context**
After seeing that NetworkManager didn't work, I manually edited /etc/network/interfaces to have the following contents:
```
# interfaces(5) file used by ifup(8) and ifdown(8)
# Use `man interfaces` to read the syntax

# Include files from /etc/network/interfaces.d:
source /etc/network/interfaces.d/*

# Make eth0 be automatically setup
auto eth0

# Set up static IPv4 for eth0
iface eth0 inet static
  address 10.24.12.51
  netmask 255.255.255.0
  gateway 10.24.12.1
```

Additionally, looking through the logs, it's trying to do stuff with "Wired connection 1", but there is no such connection:
```
ubuntu@photonvision:~$ nmcli
eth0: unmanaged
        "eth0"
        ethernet (rk_gmac-dwmac), 16:97:52:87:10:47, hw, mtu 1500

lo: unmanaged
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

Use "nmcli device show" to get complete information about known devices and
"nmcli connection show" to get an overview on active connection profiles.

Consult nmcli(1) and nmcli-examples(7) manual pages for complete usage details.
```

mcm001, 2024-02-13:
[image]
I see it later picks up eth0, but connName is still empty which means it's filtered out by https://github.com/PhotonVision/photonvision/blob/6a2d83e19ba6d0b87f8dd6ca84102805e6443fe4/photon-core/src/main/java/org/photonvision/common/networking/NetworkUtils.java#L121 . Can you reflash back to stock and upload journalctl logs of the first run of photon starting from scratch?

KangarooKoala, 2024-02-14:
Weird- After reflashing, it works now. It could be that I skipped past the NetworkManager interface dropdown at some point in initial setup and that somehow messed it up, or maybe it was something else.

Steps (for completeness):
1. Download 2024.2.4 Orange Pi 5 image
2. Turn Orange Pi 5 off
3. (Take microSD card out)
4. Use Balena Etcher (version 1.18.11) to flash the microSD card. (Side note- It would get stuck on the finishing step and wouldn't let me skip it, but I could skip the validation step. I also tried putting the SD card in after opening Balena Etcher, so that may also have been relevant.)
5. Insert microSD card into Orange Pi 5
6. Turn on Orange Pi 5
7. Plug in ethernet

Here's the log: 
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/14275172/photonvision-journalctl.txt)

I'm good with closing this as can't reproduce, though if you want me to try some more testing I could also do that.

crschardt, 2024-02-14:
Here's what is happening:

The first time it queries network manager, the ethernet link isn't fully up and it returns an empty value for connName:
```
Nov 17 06:45:09 ubuntu java[795]: [2023-11-17 06:45:09] [General - NetworkUtils] [DEBUG] Found network interfaces:
Nov 17 06:45:09 ubuntu java[795]: [NMDeviceInfo [connName=, devName=eth0, nmType=NMTYPE_ETHERNET], NMDeviceInfo [connName=, devName=lo, nmType=NMTYPE_UNKNOWN]]
```
This gets replaced by the default value of "Wired connection 1" because that is the default name on Raspberry Pi. It then tries to set up the network interface and fails:
```
Nov 17 06:45:09 photonvision java[795]: [2023-11-17 06:45:09] [General - ShellExec] [DEBUG] Executing "nmcli con mod "Wired connection 1" ipv4.method "auto" ipv6.method "disabled""
Nov 17 06:45:09 photonvision java[795]: [2023-11-17 06:45:09] [General - ShellExec] [DEBUG] Got exit code 10
Nov 17 06:45:09 photonvision java[795]: [2023-11-17 06:45:09] [General - NetworkManager] [ERROR] Exception while setting DHCP!
```
A little later, it queries the network interfaces again and this time, it gets a valid name of "netplan-eth0":
```
Nov 17 06:48:55 photonvision java[795]: [2023-11-17 06:48:55] [General - NetworkUtils] [DEBUG] Found network interfaces:
Nov 17 06:48:55 photonvision java[795]: [NMDeviceInfo [connName=netplan-eth0, devName=eth0, nmType=NMTYPE_ETHERNET], NMDeviceInfo [connName=, devName=lo, nmType=NMTYPE_UNKNOWN]]
```
however, it continues to use the "Wired connection 1" default until the user manually changes it in the Settings interface.

crschardt, 2024-10-10:
I believe this is fixed in the updated OPi image.

closed 2024-10-10 by crschardt