PhotonVision/photonvision #1240: mrcal intrinsics calculation differences between PV UI and mrcal CLI

willtoth, 2024-02-18:
We ran calibration with mrcal enabled on ~80 images. At the end of the process, the intrinsics output were

```
'intrinsics': [ 964.3192542, 968.4167198, 766.3572371, 656.0259428, -0.03374426143, 0.0003779715023, 0.0008241449289, -0.009869350879, -1.381816601e-06, -0.05084474531, 0.0007160772276, -3.0046612e-06,],
```

Running the same images using calibrationUtil.py to extract them, but with the mrcal CLI as described [here](https://mrcal.secretsauce.net/tour-initial-calibration.html), the intrinsics result is:

```
'intrinsics': [ 899.2457549, 900.0461934, 828.2229362, 663.153752, 0.007466898828, 0.01911741769, -0.0002934119232, 0.001457083979, 0.1258894663, -0.04345164244, 0.09710341804, 0.1088452785,],
```

Some data in the mrcal CLI output

```bash
# generated on 2024-02-17 22:30:20 with   /usr/bin/mrcal-calibrate-cameras --corners-cache corners.vnl --lensmodel LENSMODEL_OPENCV8 --focal 1900 --object-spacing 0.030 --object-width-n 18 --object-height-n 13 '*.png'
# RMS reprojection error: 0.2 pixels
# Worst residual (by measurement): 0.7 pixels
# Noutliers: 426 out of 18720 total points: 2.3% of the data
# calobject_warp = [-0.00067182  0.00036481]
```

I also recall the output (number of outlines) being much higher on the PV UI, but I did not capture this unfortunately.

Additionally, the calibration results by running `mrcal-show-projection-uncertainty --unset key ./cameramodel_0.cameramodel` is quite poor with the output of the python script, but shows quite good values with the output of the mrcal.

Values generated from calibrationUtils.py

[image]

mrcal CLI

[image]

**Platform (PV):**
 - x86 Intel N100 PC running Ubuntu
 - PhotonVision Version: 2024.0.4
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Firefox
 - Camera(s) Used: ov2311

**Platform (mrcal CLI):**
 - Windows 11 + WSL2

JSON file is too big for Github, download [here](https://drive.google.com/file/d/1Qo2XpKNfTK-Rq3WOm1lWBXHnJRAsbs1r/view?usp=drive_link)

mrcal results
[mrcal_camera-0.cameramodel.json](https://github.com/PhotonVision/photonvision/files/14321232/mrcal_camera-0.cameramodel.json)

mcm001, 2024-02-18:
I've seen this once before but attributed it to problems converting from photon to mrcal lensmodel file, not an underlying difference. I think the next thing to do is set breakpoints in mrcal and compare seed poses and detected corners and stuff?

willtoth, 2024-02-19:
The first time I ran this, the intrinsics calculated between the two were still different, but quite close to eachother. e.g.

mrcal
```
[ 901.7330686, 902.7952017, 799.8291256, 666.452847, 0.05673155155, -0.08460947181, -0.0009418336749, -0.0008824814115, 0.01479466453, 0.003773460448, 0.006533366048, -0.01477754318,]
```

PV
```
[900.3160750607914,0.0,799.8930541069874,0.0,901.4700410488433,667.1849100287695,0.0,0.0,1.0]
```

The camera and lens itself is the same model between this and the one in the original message. The values are pretty close. Either way, for the intrinsics calculation, do you expect the mrcal CLI or the results from the PV run to be more accurate?

I won't have time to help debug for the next 2 or so weeks as we prepare for our first event, but I can probably dig into it a bit after that.

randomstring, 2024-02-19:
I too am seeing differences between the PV output and what mrcal computes. When I map the uncertainty of the PV calibration I get an all yellow graph like the one linked above. 

running calibration like this:  (8x8 grid with 1" spacing, correct me if I'm getting the arguments wrong.)

```
mrcal-calibrate-cameras --corners-cache corners.vnl --lensmodel LENSMODEL_OPENCV8  --focal 1900 --object-spacing .02540125    --object-width-n 7  '*.png'
```

I get a more reasonable map of error, with a sweet spot in the middle in the 0.2 error rage with more error around the edges.

mcm001, 2024-02-20:
For continuity , are the intrinsics you get different? Or is only the graph.

randomstring, 2024-02-21:
Yes, intrinsics were slightly different, mostly values just a few percent different:  685.66 -> 686.27, etc.

dkogan, 2024-02-28:
Hello. I'm the mrcal author. Feel free to ask me for help about things mrcal itself is doing; use the mrcal bug tracker maybe. For photonvision problems, I cannot say much, but I'm happy to help @mcm001 or whoever else debug stuff.

mcm001, 2024-02-28:
I think that debugging this is probably going to require I sit down with a debugger and compare inputs to mrcal line-by-line against your implementation again. 

Does mrcal-calibrate-cameras by default perform a calibration refinement? Right now, our calibration is just a single call to `mrcal_optimize` [(see here)](https://github.com/PhotonVision/mrcal-java/blob/da8c4e9bf7205a0f0e20f0cbfc02f8ec92ea2079/src/mrcal_wrapper.cpp#L216C16-L216C30). And we generate seed poses using [solvepnp_epnp](https://github.com/PhotonVision/mrcal-java/blob/da8c4e9bf7205a0f0e20f0cbfc02f8ec92ea2079/src/mrcal_wrapper.cpp#L348).

mcm001, 2024-02-28:
[camera-0_RESOLVE_WITH_MRCAL_CALIBRATE_CAMERAS.zip](https://github.com/PhotonVision/photonvision/files/14440442/camera-0_RESOLVE_WITH_MRCAL_CALIBRATE_CAMERAS.zip)

Here's the cameramodel I got from using `mrcal-calibrate-cameras`, which again matches the results you got using the mrcal CLI. I then wrote a little unit test to use your calibration JSON (attached in your comment above) and try calibrating. This is the output I see.
```java
    @Test
    public void testWillBoard() throws IOException {
        var cal = JacksonUtils.deserialize(Path.of("/home/matt/mrcal_debug_tmp/pv_calibration_1600x1200.json"), 
            CameraCalibrationCoefficients.class);
        // System.out.println(cal);

        var corners = cal.observations.stream().map(it->{
                var mat = new MatOfPoint2f();
                mat.fromList(it.locationInImageSpace);
                return mat;
            }).collect(Collectors.toList());

        var result = MrCalJNI.calibrateCamera(
            corners,
            18, 13, 0.03, 
            1600, 1200, 1200
        );
        System.out.println(result);
    }
```
Calibrate3dPipeTest > testWillBoard() STANDARD_OUT
    CameraCalibrationCoefficients [resolution=1600x1200, cameraIntrinsics=JsonMat [rows=3, cols=3, type=6, data=[964.3192542051779, 0.0, 766.3572370964357, 0.0, 968.4167197906467, 656.0259427916941, 0.0, 0.0, 1.0], wrappedMat=Mat [ 3*3*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7fe345180730, dataAddr=0x7fe345180f40 ], wpilibMat=null, wrappedMatOfDouble=null], distCoeffs=JsonMat [rows=1, cols=8, type=6, data=[-0.03374426143309665, 3.779715023416348E-4, 8.241449288523903E-4, -0.009869350879493375, -1.3818166014360312E-6, -0.050844745308818114, 7.160772275740329E-4, -3.0046612000201336E-6], wrappedMat=Mat [ 1*8*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7fe34510dc80, dataAddr=0x7fe34519b680 ], wpilibMat=null, wrappedMatOfDouble=Mat [ 1*8*CV_64FC1, isCont=true, isSubmat=false, nativeObj=0x7fe34510dc10, dataAddr=0x7fe3451ab080 ]], observationslen=80, calobjectWarp=[-2.25221267024838E-4, 7.336652791935415E-4], intrinsicsArr=[964.3192542051779, 0.0, 766.3572370964357, 0.0, 968.4167197906467, 656.0259427916941, 0.0, 0.0, 1.0], distCoeffsArr=[-0.03374426143309665, 3.779715023416348E-4, 8.241449288523903E-4, -0.009869350879493375, -1.3818166014360312E-6]]

libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114: CHOLMOD warning:
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  not positive definite.
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  file: ../Cholesky/t_cholmod_rowfac.c
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  line: 430
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114: 

Watching 173 directories to track changes
/home/runner/work/mrcal-java/mrcal-java/mrcal/mrcal.c(5540): Threw out some outliers. New count = 425/37440 (1.1%). Going again
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114: CHOLMOD warning:
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  not positive definite.
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  file: ../Cholesky/t_cholmod_rowfac.c
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114:  line: 430
libdogleg at /home/runner/work/mrcal-java/mrcal-java/libdogleg/dogleg.c:1114: 

Gradle Test Executor 13 finished executing tests.

> Task :photon-core:testHeadless
    MrCalResult [success=true, intrinsics=[815.2444290805902, 702.5299171031456, 783.7045709624529, 598.5380702840318, 16.022877352468576, -11.617096993914174, -0.012759261238650944, -0.16064034039996528, -6.856871701238218, 11.279315510816447, -8.68659166855614, -4.641638738271045], rms_error=48.3446650298785, warp_x=0.005452025977464822, warp_y=0.0011008387439970853, Noutliers=425]
```

dkogan, 2024-03-02:
> Does mrcal-calibrate-cameras by default perform a calibration
> refinement? Right now, our calibration is just a single call to
> mrcal_optimize. And we generate seed poses using solvepnp_epnp.

mrcal-calibrate-cameras does this also: there's an incremental seeing
and solving routine. It's implemented in Python, which I guess means you
can't use it directly. The seed routine is here:

  https://github.com/dkogan/mrcal/blob/90818fdbb6ef422c90b644690d5b14add3a42724/mrcal/calibration.py#L1360

Look at mrcal-calibrate-cameras to see how that's called.

dkogan, 2024-03-02:
I looked at the model in the .zip file and the model in the smaller
.json file above (mrcal models aren't json: they're python dicts; so
they can have comments, and such). Both those models show the "small"
uncertainties.

Is this bug report summarized as "photonvision produces results with
alarmingly-high reported uncertainties"? If so, I'd need a model that
shows such high uncertainties to tell you what's going on. Are the
high-uncertainty produced by running
mrcal-show-projection-uncertainties, or is that wrapper inside
photonvision somehow? I'd need some way to reproduce whatever failure
you're seeing.

Similarly for the "not-positive-definite" messages you're seeing: if I
reoptimize either for the models I have, I don't see this. The most
common cause of these messages is described here:

  https://mrcal.secretsauce.net/how-to-calibrate.html#orgea10dd0

mcm001, 2024-03-03:
Yeah, that summery is accurate. For a little more context, Photon uses JSON files for camera calibration, and these jsons contain the images, board corner observations, and other optimization inputs. We also provide a python script to convert this json to a mrcal cameramodel, and try to [convert out optimization inputs to mrcal-land, ](https://github.com/PhotonVision/photonvision/blob/7cec1413410c8053dafd8ecb57095f47469c758c/devTools/calibrationUtils.py#L144C1-L184C6)so we can use your tools like `mrcal-show-projection-uncertainty`. The core bug report here is that when a calibration performed using PhotonVision is exported and converted to a mrcal cameramodel, the projection uncertainty shown using `mrcal-show-projection-uncertainties` is alarmingly high. 

A bug could exist in our export/conversion script, or we could truly have a calibration that's wildly wrong.

mcm001, 2024-03-03:
> mrcal-calibrate-cameras does this also: there's an incremental seeing and solving routine. It's implemented in Python, which I guess means you can't use it directly. The seed routine is here: [...]

right now we just jump straight into a full solve optimizing everything simultaniously, which probably doesn't help. I'll try changing our code to properly do the same incremental optimization process you linked and see how that changes our results.

mcm001, 2024-03-03:
I've started prototyping this code over [in this branch](https://github.com/PhotonVision/mrcal-java/compare/main...incremental-solve), but i'm still getting weird results. Needs more investigation

dkogan, 2024-03-03:
> The core bug report here is that when a calibration performed using
> PhotonVision is exported and converted to a mrcal cameramodel, the
> projection uncertainty shown using mrcal-show-projection-uncertainties
> is alarmingly high.

OK. Can I get a copy of one of those .cameramodel files that produce a
very high mrcal-show-projection-uncertainty result? I believe all the
ones that were linked in this report have reported low uncertainties

> right now we just jump straight into a full solve optimizing
> everything simultaniously, which probably doesn't help. I'll try
> changing our code to properly do the same incremental optimization
> process you linked and see how that changes our results.

I won't claim that the seeding sequence I use currently is perfect, but
it has been tested thoroughly, and does work ok. Is there really no way
to use the Python code directly in photovision? If doing a rewrite in
another language, it'll take a while (lots of testing and fixing things)
to get to the same level of performance, and it feels like effort is
being duplicated. You can embed Python in C like this:

  https://docs.python.org/3/extending/embedding.html

Worth thinking about?

mcm001, 2024-03-03:
The main reason I'm hesitant to pursue that is it would require we install python3 and mrcal in our image. This would add ~450MB of packages to our image (at least for the `mrcal` package, version 2.4.1-1jammy1). Just python3 itself adds ~25MB, so I assume most of this extra size is from other dependencies the `mrcal` package pulls in? 

If there's any way to get the size of the package down while still being able to wrap python, that could be something we consider for 2025. You're right that I'm not a fan of how much API I'm having to duplicate. I'd just also like to be cognizant of image size.

dkogan, 2024-03-03:
You're talking about the debian packages here, and the "450MB" number
comes from what APT tells you in will need when you ask to "apt install
python3-mrcal"? If not, where does "450MB" come from? Presumably you're
already installing "libmrcal4" and "libmrcal-dev", and you're looking at
the EXTRA space "python3-mrcal" will require?

If so, can you get some more detailed accounting, to see which packages
specifically are responsible for the ballooning size? mrcal itself and
things it directly uses aren't nearly this fat, but there could be
optional things, or things that dependencies optionally need.

I can help you get the detailed numbers, if you tell me exactly where
you get the "450MB" number.

mcm001, 2024-03-04:
Sure. Starting from a blank [ubuntu:22.04](https://hub.docker.com/layers/library/ubuntu/22.04/images/sha256-81bba8d1dde7fc1883b6e95cd46d6c9f4874374f2b360c8db82620b33f6b5ca1?context=explore) docker image, I'm just using the size reported by apt (eg, `After this operation, 264 MB of additional disk space will be used.

- `apt-get install --yes openjdk-17-jre-headless` -- 260 MB 
- `apt-get install --yes libcholmod3 liblapack3 libsuitesparseconfig5` -- 14 MB. These packages are what mrcal-java links to. Mrcal-java is a build that contains both mrcal and our JNI in the same so. mrcal-java shared libraries live inside of the photonvision JAR file we [distribute](https://maven.photonvision.org/#artifact-details-download-content/org.photonvision/photon-mrcal-jni/dev-v2024.0.0-9-gbec18e3), and on linux that's 107kb compressed in a jar. 
- `libmrcal4` -- 24 MB. I don't think it makes sense to install libmrcal-dev on user systems right, since they don't need to be compiling code?
<details>
<summary>Detailed command output</summary>
<br>
```
root@ab408bde2eb4:/# apt show libmrcal4
Package: libmrcal4
Version: 2.4.1-1jammy1
Priority: optional
Section: libs
Source: mrcal
Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Installed-Size: 310 kB
Depends: libc6 (>= 2.33), libdogleg2 (>= 0.15.3), libfreeimage3 (>= 3.18.0+ds2)
Homepage: http://mrcal.secretsauce.net
Download-Size: 112 kB
APT-Sources: http://mrcal.secretsauce.net/packages/jammy/public jammy/main amd64 Packages
Description: Calibration and SFM library
 A photogrammetry toolkit that provides improved methods for making and using
 camera models (calibration, tracking, mapping, etc). The focus is on
 calibration and uncertainty propagation to produce the most accurate models
 possible. This package provides the runtime C libraries

root@ab408bde2eb4:/# apt install libmrcal4
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libamd2 libblas3 libcamd2 libccolamd2 libcholmod3 libcolamd2 libdeflate0 libdogleg2 libfreeimage3 libgfortran5 libgomp1 libilmbase25 libjbig0 libjxr0 liblapack3 libmetis5 libopenexr25 libopenjp2-7 libquadmath0 libraw20 libsuitesparseconfig5 libtiff5 libwebp7 libwebpmux3
The following NEW packages will be installed:
  libamd2 libblas3 libcamd2 libccolamd2 libcholmod3 libcolamd2 libdeflate0 libdogleg2 libfreeimage3 libgfortran5 libgomp1 libilmbase25 libjbig0 libjxr0 liblapack3 libmetis5 libmrcal4 libopenexr25 libopenjp2-7 libquadmath0 libraw20 libsuitesparseconfig5 libtiff5 libwebp7
  libwebpmux3
0 upgraded, 25 newly installed, 0 to remove and 0 not upgraded.
Need to get 7082 kB of archives.
After this operation, 23.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] ^C
root@ab408bde2eb4:/#
```
</details>
- `python3-mrcal`: after installing libmrcal4, an additional 1615 MB
<details>
<summary>apt install output</summary>
<br>
```
root@ab408bde2eb4:/# apt show python3-mrcal
Package: python3-mrcal
Version: 2.4.1-1jammy1
Priority: optional
Section: python
Source: mrcal
Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Installed-Size: 1555 kB
Provides: python3.10-mrcal
Depends: libc6 (>= 2.14), libmrcal4 (= 2.4.1-1jammy1), libsuitesparseconfig5 (>= 1:5.0.0), python3-numpy (>= 1:1.20.0), python3-numpy-abi9, python3 (<< 3.11), python3 (>= 3.10~), python3:any, python3-gnuplotlib (>= 0.38), python3-numpysane (>= 0.35), python3-opencv, python3-scipy (>= 0.18), python3-shapely, python3-ipython, python3-matplotlib, python3-yaml
Homepage: http://mrcal.secretsauce.net
Download-Size: 267 kB
APT-Sources: http://mrcal.secretsauce.net/packages/jammy/public jammy/main amd64 Packages
Description: Calibration and SFM library

N: There is 1 additional record. Please use the '-a' switch to see it
root@ab408bde2eb4:/# apt show python3-mrca^C
root@ab408bde2eb4:/# apt install python3-mrcal
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
[... 44 lines cut ...]
  libibverbs1 libice6 libigdgmm12 libimagequant0 libinput-bin libinput10 libip4tc2 libisl23 libitm1 libjs-jquery libjs-jquery-ui libjs-sphinxdoc libjs-underscore libjson-c5 libjsoncpp25 libkmlbase1 libkmldom1 libkmlengine1 libkmod2 liblbfgsb0 libldap-2.5-0 libldap-common liblept5
  libllvm15 liblsan0 libltdl7 liblua5.4-0 libmagic-mgc libmagic1 libmd0 libmd4c0 libmfx1 libminizip1 libmp3lame0 libmpc3 libmpfr6 libmpg123-0 libmtdev1 libmysqlclient21 libnetcdf19 libnghttp2-14 libnl-3-200 libnl-route-3-200 libnorm1 libnotify4 libnsl-dev libnss-systemd libnuma1
  libodbc2 libodbcinst2 libogdi4.1 libogg0 libopenblas-dev libopenblas-pthread-dev libopenblas0 libopenblas0-pthread libopencv-calib3d4.5d libopencv-contrib4.5d libopencv-core4.5d libopencv-dnn4.5d libopencv-features2d4.5d libopencv-flann4.5d libopencv-highgui4.5d
  libopencv-imgcodecs4.5d libopencv-imgproc4.5d libopencv-ml4.5d libopencv-objdetect4.5d libopencv-photo4.5d libopencv-shape4.5d libopencv-stitching4.5d libopencv-video4.5d libopencv-videoio4.5d libopencv-viz4.5d libopengl0 libopenmpi3 libopenmpt0 libopus0 liborc-0.4-0 libpam-cap
  libpam-systemd libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpciaccess0 libpcre2-16-0 libperl5.34 libpgm-5.3-0 libpixman-1-0 libpmix2 libpoppler118 libpq5 libproj22 libprotobuf23 libpsl5 libpsm-infinipath1 l [... cut ...]
  python3-unicodedata2 python3-wcwidth python3-webencodings python3-yaml python3.10 python3.10-dev python3.10-minimal qt5-gtk-platformtheme qttranslations5-l10n rpcsvc-proto session-migration systemd systemd-sysv systemd-timesyncd tk8.6-blt2.5 tzdata ubuntu-mono unicode-data
  unixodbc-common va-driver-all vdpau-driver-all x11-common xkb-data xz-utils zlib1g-dev
0 upgraded, 472 newly installed, 0 to remove and 0 not upgraded.
Need to get 362 MB of archives.
After this operation, 1615 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
root@ab408bde2eb4:/#
```
</details>

mcm001, 2024-03-04:
sorry, github formatting was messed up initially for my above comment. should be resolved now.

dkogan, 2024-03-04:
OK. I just poked at it a bit. If I ask for the size of everything needed to install `python3-mrcal` that isn't also needed for `libmrcal4`, it says there's 1159MB extra:

```
for pkg (`join -v1 <(apt-rdepends python3-mrcal | grep -v '^ ' | sort) <(apt-rdepends libmrcal4 | grep -v '^ ' | sort)`) { echo $pkg/bookworm } | xargs apt-cache show | awk '/^Package:/ {p=$2} /^Installed-Size:/ { s+=$2 } END {print s/1e3}'
```

It's a bit different from your other numbers for uninteresting reasons; hopefully much of the reasoning would be the same either way. These are installed sizes. I looked at it a bit, and it looks like 735MB of that is `python3-opencv`. This package definitely isn't as fine-grained as it could be. So today, if you want to be able to `import cv2` in Python, you need 735MB of space.

Of the remaining stuff, I THINK much of it is from installing the `gnuplot-qt` flavor of gnuplot, which pulls in a lot of stuff. There are other flavors of that, which would be lighter. So what do you see if you

```
apt install --no-install-recommends python3-mrcal gnuplot-x11
```

Hopefully that will give you a better number. Although the biggest hog (`python3-opencv`) will still be there.

I'm still not sure why we care about this though. Disks today are very large (which is why these packages are fatter than you want: nobody has cared enough to trim them). You are using a very common OS with very common packages. Why do you even need an image? You can use something like `equivs` to build a (tiny) meta-package that depends on the things that you need (let's call it `photon-vision-deps`). Then the instructions are:

- Install base OS. Could be natively or in a chroot of in VM via an image. Whatever; doesn't matter.
- Install `photon-vision-deps`

Then even if `photon-vision-deps` contains some fat thing like `python3-opencv`, it isn't something you think about. And the students won't either: it's a fraction the size of a netflix movie.

And as an added bonus, having Python and opencv available is something the students should find useful.

Sorry if that's not completely what you wanted to hear. If you had infinite time and motivation, you could improve the `python3-opencv` packaging to make it more granular, to allow installing only the pieces we need. But that's not obviously a good use of anybody's time, which is why nobody has done it.

If you need help with equivs or setting up an APT server with the extra packages you need or whatever, feel free to ask. I do this all the time.

mcm001, 2024-03-04:
Robots are generally an offline environment, which drives the need for a completely offline solution. For competition, we also need to disable wireless, which forces you to somehow find a wired internet port at your school/workshop. I agree that in a perfect world, you'd just plug your coprocessor into your school LAN and download some packages, but that's outside the reach of a good chunk of our users.

The concern about total disk space is to keep image download/flashing speed from bloating too much, and also just proactively trying to not hit Github's download size limit. I'm just not sure if the value prop is there for increasing the size of our image this much is there yet. 

From a fresh docker image of ubuntu:22.04, I see:

<details><summary>apt install --no-install-recommends gnuplot-x11</summary>

```
root@2d3cc328ef2c:/# apt install gnuplot-x11
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  adwaita-icon-theme aglfn at-spi2-core dbus dbus-user-session dconf-gsettings-backend dconf-service dmsetup fontconfig fontconfig-config fonts-dejavu-core gir1.2-glib-2.0 gnuplot-data gsettings-desktop-schemas
  gtk-update-icon-cache hicolor-icon-theme humanity-icon-theme libapparmor1 libargon2-1 libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3 libbrotli1
  libbsd0 libcairo-gobject2 libcairo2 libcolord2 libcryptsetup12 libcups2 libdatrie1 libdbus-1-3 libdconf1 libdeflate0 libdevmapper1.02.1 libdrm-amdgpu1 libdrm-common libdrm-intel1 libdrm-nouveau2 libdrm-radeon1
  libdrm2 libedit2 libelf1 libepoxy0 libfontconfig1 libfreetype6 libfribidi0 libgd3 libgdk-pixbuf-2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgirepository-1.0-1 libgl1 libgl1-amber-dri libgl1-mesa-dri
  libglapi-mesa libglib2.0-0 libglib2.0-data libglvnd0 libglx-mesa0 libglx0 libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libharfbuzz0b libice6 libicu70 libip4tc2 libjbig0 libjpeg-turbo8 libjpeg8 libjson-c5
  libkmod2 liblcms2-2 libllvm15 liblua5.4-0 libmd0 libnotify4 libnss-systemd libpam-systemd libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpciaccess0 libpixman-1-0 libpng16-16 libpython3-stdlib librsvg2-2
  librsvg2-common libsensors-config libsensors5 libsm6 libthai-data libthai0 libtiff5 libwayland-client0 libwayland-cursor0 libwayland-egl1 libwebp7 libwxbase3.0-0v5 libwxgtk3.0-gtk3-0v5 libx11-6 libx11-data
  libx11-xcb1 libxau6 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcb1 libxcomposite1 libxcursor1 libxdamage1 libxdmcp6 libxext6
  libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxml2 libxpm4 libxrandr2 libxrender1 libxshmfence1 libxtst6 libxxf86vm1 networkd-dispatcher python3 python3-dbus python3-gi python3-minimal python3.10
  python3.10-minimal session-migration shared-mime-info systemd systemd-sysv systemd-timesyncd ubuntu-mono ucf x11-common xdg-user-dirs xkb-data
Suggested packages:
  gnuplot-doc colord cups-common libgd-tools gvfs liblcms2-utils gnome-shell | notification-daemon pciutils librsvg2-bin lm-sensors iw | wireless-tools python3-doc python3-tk python3-venv python-dbus-doc
  python3.10-venv python3.10-doc binutils binfmt-support systemd-container libfido2-1 libtss2-esys-3.0.2-0 libtss2-mu0 libtss2-rc0 policykit-1
The following NEW packages will be installed:
  adwaita-icon-theme aglfn at-spi2-core dbus dbus-user-session dconf-gsettings-backend dconf-service dmsetup fontconfig fontconfig-config fonts-dejavu-core gir1.2-glib-2.0 gnuplot-data gnuplot-x11
  gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme humanity-icon-theme libapparmor1 libargon2-1 libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data
  libavahi-common3 libbrotli1 libbsd0 libcairo-gobject2 libcairo2 libcolord2 libcryptsetup12 libcups2 libdatrie1 libdbus-1-3 libdconf1 libdeflate0 libdevmapper1.02.1 libdrm-amdgpu1 libdrm-common libdrm-intel1
  libdrm-nouveau2 libdrm-radeon1 libdrm2 libedit2 libelf1 libepoxy0 libfontconfig1 libfreetype6 libfribidi0 libgd3 libgdk-pixbuf-2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgirepository-1.0-1 libgl1
  libgl1-amber-dri libgl1-mesa-dri libglapi-mesa libglib2.0-0 libglib2.0-data libglvnd0 libglx-mesa0 libglx0 libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libharfbuzz0b libice6 libicu70 libip4tc2 libjbig0
  libjpeg-turbo8 libjpeg8 libjson-c5 libkmod2 liblcms2-2 libllvm15 liblua5.4-0 libmd0 libnotify4 libnss-systemd libpam-systemd libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpciaccess0 libpixman-1-0
  libpng16-16 libpython3-stdlib librsvg2-2 librsvg2-common libsensors-config libsensors5 libsm6 libthai-data libthai0 libtiff5 libwayland-client0 libwayland-cursor0 libwayland-egl1 libwebp7 libwxbase3.0-0v5
  libwxgtk3.0-gtk3-0v5 libx11-6 libx11-data libx11-xcb1 libxau6 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 libxcb1 libxcomposite1
  libxcursor1 libxdamage1 libxdmcp6 libxext6 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxml2 libxpm4 libxrandr2 libxrender1 libxshmfence1 libxtst6 libxxf86vm1 networkd-dispatcher python3 python3-dbus python3-gi
[... cut ...]
  python3-prompt-toolkit python3-ptyprocess python3-pygments python3-pyparsing python3-pythran python3-scipy python3-shapely python3-six python3-sympy python3-tk python3-traitlets python3-tz python3-ufolib2
  python3-unicodedata2 python3-wcwidth python3.10 python3.10-dev python3.10-minimal rpcsvc-proto shared-mime-info systemd systemd-sysv tk8.6-blt2.5 tzdata ubuntu-mono ucf unicode-data unixodbc-common x11-common
  xkb-data zlib1g-dev
0 upgraded, 426 newly installed, 0 to remove and 0 not upgraded.
Need to get 342 MB of archives.
After this operation, 1497 MB of additional disk space will be used.
```

</details>

dkogan, 2024-03-04:
> Robots are generally an offline environment, which drives the need for
> a completely offline solution. For competition, we also need to
> disable wireless, which forces you to somehow find a wired [...]

I don't follow. Currently you download an image from the internet and
use it to initialize an OS with some packages; then you can be offline.
How's that different from downloading some packages from the internet
and then going offline?

> The concern about total disk space is to keep image download/flashing
> speed from bloating too much, and also just proactively trying to not
> hit Github's download size limit. I'm just not sure if the value prop
> is there for increasing the size of our image this much is there yet.

I don't know your system at all, so I'm going to stop proposing things.
If you want to have a separate conversation about Debian packages and
apt servers and managing installs and whatnot, we can do that separately
from this issue. I'm a Debian Developer, and have managed many such
things previously.

> From a fresh docker image of ubuntu:22.04, I see:
>
> ...
> After this operation, 1497 MB of additional disk space will be used.

OK, so we saved ~ 100MB by picking a different gnuplot? That's not
nothing, but it's also not everything either.

mcm001, 2024-03-04:
Another thing I'll throw onto the pile for consideration. Our current JNI setup doesn't require Windows users install any additional programs to make Photon work. If we go the path of directly calling mrcal-python, that won't be true anymore. I think that whatever solution we end up at, making sure it work painlessly under windows is a hard requirement.

dkogan, 2024-03-04:
Matt ***@***.***> writes:

> they tend to be completely headless setups -- is it worth trying to
> avoid installing gnuplot/other GUI tools altogether?

Some things are still useful: you can x-forward gnuplot for instance.
But the gnuplot packages have flavors of different fatness (because
those packages were built when disk space was more at a premium). You
certainly don't need most of opencv, but it would take some work to
disentangle that

> Photonvision also ships with opencv (not installed as a package, but
> as just shared libraries from our opencv fork here embedded in the
> JAR, so it seems duplicative to also install opencv packages assuming
> mrcal could work with opencv 4.8.0.

That's on photonvision for doing something weird. It should be
installing and using the system packages.

> And you only need a couple modules from opencv for a vanilla monocular
> camera calibration right? (I saw solvePNP used, but I think that was
> it). I see used:
>
> * cv2.findContours
> * cv2.contourArea
> * cv2.solvePNP

That sounds right.

> I also noticed that python3-opencv is using packages named
> libopencv-[MODULE]4.5d -- does this mean that they're packages with
> debug symbols?

The meaning of the "d" is a question for whoever wrote the opencv build
system. The debian packaging strips out the debug symbols into separate
packages (called whatever-dbgsym). All the package sizes we've been
looking at do not include the debug symbols.

> I think if we can get opencv trimmed down a bit, this could be a
> viable option to pursue once the season wraps up

Let's talk separately from this issue later, when you actually want to
undertake bigger surgery.

dkogan, 2024-03-05:
Awesome! Glad you found that. Let me know if you hit any other issues
that I can help with.

Juniormunk, 2024-10-09:
Fix was merged.

closed 2024-10-09 by Juniormunk