PhotonVision/photonvision #1242: Undistort pitch/yaw for colored shape targets

PascalSkylake, 2024-02-18:
I've been trying to use photonvision for detecting game pieces, but there is significant error when getting pitch and yaw because of lens distortion, making the absolute position of the game pieces change as the camera rotates.

I left a comment about this a little while ago on an issue from 2022, but that issue was only somewhat related.

mcm001, 2024-02-18:
Is this for colored shape or also for ML results? Either way yeah agreed. What lens FOV do you have?

PascalSkylake, 2024-02-19:
I'm asking specifically for colored shape, but I imagine the process wouldn't be too different? Off the top of my head the camera is 65 degrees hfov, I can send actual calibration results tomorrow if that'd help, but the issue is present on every camera I've tried.

mcm001, 2024-02-21:
https://github.com/PhotonVision/photonvision/pull/1250 has the start of this. I'm scared to touch critical code like this so late in the season, though.

Juniormunk, 2024-10-09:
#1250 was merged closing.

closed 2024-10-09 by Juniormunk