PhotonVision/photonvision #1248: Quirky Camera detection doesn't always set exposure range

TheTripleV, 2024-02-20:
Sometimes, it seems like the quirky camera exposure range adjustment isn't applied and the scaling is wrong in the UI.
After a reboot, the scaling is back to normal.

When this happens, almost all UI exposure values will result in an overexposed near white frame, and a values of 0-0.5 cause large jumps in exposure.

I'll try and find/capture logs for when this happens.

**Platform:**
 - Hardware Platform: Opi5
 - PhotonVision Version: main
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): chrome
 - Camera(s) Used: ov2311 x 2

Breadspeed1, 2024-02-24:
Did you have to restart the coprocessor or just the instance of photonvision? We are using two Opi5s with various assortments of ov2311s and ov9782s and it seems that only the coprocessor with the 2311 has the exposure bug, and it affects the 9782 on there as well. I haven't had any luck with restarting pv, but occasionally toggling autoexposure a few times fixes it

TheTripleV, 2024-02-25:
It happens on our ov2311, which should appear is ov9281 iirc.

Note ov9782 is not supported by photonvision for setting exposure properly. See #1204

Breadspeed1, 2024-02-25:
Oh, on our other opi exposure setting works fine with the 9782s, but it seems like the 2311 messes up the ability to set exposure at all on the opi that it's connected to, even with other cameras. We've just decided to use autoexposure and it seems to work alright. Thank you for your quick response!

DrewTheRat, 2024-03-18:
This seems to be an issue with the LifeCam HD3000. If I change the brightness even slightly, the image gets dramatically brighter, and moving back to the previous value does not return the image to what it was. In addition, moving the exposure immediately darkens the image. See attached video.

https://github.com/PhotonVision/photonvision/assets/6615787/bf2a2f0d-362c-4ec3-845a-233514e6d7b6

mcm001, 2024-03-18:
That's a lifecam special I think, and unrelated to this issue

mcm001, 2024-11-02:
Likely OBE via Arducam quirk setting and raw exposure range?

closed 2024-11-02 by mcm001