PhotonVision/photonvision #1252: v2024.2.7 unable to boot on orange pi 5

brothersw, 2024-02-22:
**Describe the bug**
Orange pi 5 fails to boot with the latest version of photonvision. No green status light, only the red light turns on.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash SD card with latest version on latest balena etcher install
2. Plug in & network radio and orange pi

**Platform:**
 - Hardware Platform: Orange Pi 5
 - Network Configuration: tested directly to FRC radio or with a network switch
 - PhotonVision Version: 2024.2.7
 - Browser: Firefox and ungoogled chromium
 - Camera(s) Used: Arducam

**Additional context**
No HDMI or network presence. I do not have a TTL adapter to read debug. I am additionally able to boot into other operating systems, including armbian CLI. Nmap shows only my device and the radio, checking across the expected ports.

brothersw, 2024-02-22:
User archduketim in the photonvision discord is having the same troubles on the latest version

mcm001, 2024-02-22:
Vcubed did not see this with the latest CI run on master before 2024.2.7. Does that image work for y'all?

Your balena etcher version and OS can sometimes cause weird problems but you said you had flashed images in the past OK. Might still be worth confirming.

srimanachanta, 2024-03-06:
Can confirm that 2024.2.7 works on Orange Pi 5

brothersw, 2024-03-06:
Offline updating worked for me but not the boot image. I am able to use the boot iname of the prior update and then offline update.

gartaud, 2024-03-11:
Did you try 2024.2.8 by any chance?

brothersw, 2024-03-12:
No, I have not tried flashing a new install image. Offline updating has worked for all photovision updates for me so far, so that is what I have been doing.

crschardt, 2024-05-04:
Is this still a problem for you with the version 2024.3.1?

gartaud, 2024-05-04:
Not OP. Ok for my team.

srimanachanta, 2024-05-14:
Closed due to no-response from OP.

closed 2024-05-14 by srimanachanta