PhotonVision/photonvision #1253: v2024.2.7 -> Limelight3 does not save any configuration settings(web ui)

fc4furret, 2024-02-23:
**Describe the bug**
Upon restarting, photonvision web ui does not save any camera name or configuration settings. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Limelight 3 through usb-c with balena etcher and new limelight3 image. 
2. Turn on computer and robot, connect to limelight3 through web-interface. 
3. Configure robot name, set pipeline to apriltag, set robot camera name to anything. 
4. Powercycle robot, connect to camera through web-interface again. 
5. See error. 

**Screenshots / Videos**
If applicable, add screenshots to help explain your problem. Additionally, provide journalctl logs and settings zip export. If your issue is regarding the web dashboard, please provide screenshots and the output of the browser console.

Please let me know of any useful keywords to use with journalctl. 
https://i.imgur.com/gzrpQLM.png <- Image of web ui after rebooting after changing config settings once. 
https://i.imgur.com/yeUo9SK.png <- Static IP works, and that config is saved. 

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Limelight 3
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Network Switch to Limelight Ethernet. 
 - PhotonVision Version: v2024.2.7
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Firefox(123.0) Edge(121.0.2277.128) 
 - Camera(s) Used: Limelight3

**Additional context**
Add any other context about the problem here.

fc4furret, 2024-02-23:
Journalctl Logs: 
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/14390696/photonvision-journalctl.txt)

Looks like it's not saving configs to the sql? : https://imgur.com/P09wr4W

fc4furret, 2024-02-24:
Journalctl logs - 
[photonvision-journalctl.txt](https://github.com/PhotonVision/photonvision/files/14391372/photonvision-journalctl.txt)

mcm001, 2024-02-25:
Does this happen with 2024.2.5, too? Is it in any way possible that the image is somehow broken? We've had a couple reports of weird behavior on 2024.2.7 now, which is a bit concerning

fc4furret, 2024-02-25:
The issue is resolved on 2024.2.5. I reinstalled the image and jar and tried both reflashing and offline update from 2024.2.5. The issue still remained. Seems like an issue with 2024.2.7 specifically.

mcm001, 2024-02-25:
Odd thanks for testing. I decided to re run our CI pipeline on the 2024.2.7 release so maybe the new one will actually work, once that runs?

jasone, 2024-02-26:
I'm going to pile on here with what appears to be the same problem, but on a Pi 4B using an OV9281 camera. I successfully configured a static IP address, calibrated the camera for an AprilTag pipeline, all on the first boot. On the second boot, the camera config appears not to have matched what was in the database, so there was an unused "OV9281" and a new "OV9281 (1)". Multiple reboots show that previous camera configs never match; I got up to "OV9281 (1) (2) (3) (4) (5)" during testing.

Any suggestions for workarounds? Quick fix? If not, how can I use 2024.2.5 in the context of a newly brought-up 2024 FRC robot project? (I can't find downloads for older versions.)

mcm001, 2024-02-26:
I'm in mobile still, but I think this has happened enough times that something seems to be up for sure. I'll look closer at logs, but do you see any funny prints from the VisionSourceManager matching process that indicate what's up?

jasone, 2024-02-26:
I don't see any significant difference between the first and second boot in the log messages, aside from this line:

```
Dec 11 05:05:36 camera2 java[448]: [2023-12-11 05:05:36] [Camera - VisionSourceManager] [DEBUG] Trying to match 1 unmatched configs...
```

Every other line leading up to assigning a new camera name looks the same.

mcm001, 2024-02-26:
> Here is the settings export: [photonvision-settings-export.zip](https://github.com/PhotonVision/photonvision/files/14394051/photonvision-settings-export.zip)
> 
> 
> 
> Yea it looks like there is camera config being saved to the filesystem. The SQLlite seems to know the configs existed at some point though?
> 
> 
> 
> ```
> 
> tablecamerascamerasCREATE TABLE cameras (
> 
>  unique_name TINYTEXT PRIMARY KEY,
> 
>  config_json text NOT NULL,
> 
>  drivermode_json text NOT NULL,
> 
>  pipeline_jsons mediumtext NOT NULL
> 
> , otherpaths_json TEXT NOT NULL DEFAULT '[]')-A
> 
> ``` 

Yeah I agree. The database has cameras they just aren't getting matched. I think we found a array if condition  that might cause this lol[image]

fc4furret, 2024-02-28:
The new release fixed our issue with the Limelight! Resolved as closed.

closed 2024-02-28 by fc4furret