PhotonVision/photonvision #1254: No library update for v2024.2.7

brhea, 2024-02-23:
Running v2024.2.6 on orange pi 5
Downloaded and updated to v2024.2.7 on orange pi
Rebooted robot and got the fatal error message that the library did not match the version of photonvision.
Tried to update the vscode library online and none was found. 
Reverted the software on the orange pi to v2024.2.6 and error went away.

library updates have previously worked with no issue.

mcm001, 2024-02-24:
https://maven.photonvision.org/repository/internal/org/photonvision/photonlib-json/1.0/photonlib-json-1.0.json correctly shows 2024.2.7 for me....

lorelaib182, 2024-02-24:
I tried the update again today and the update to the library worked properly. could have been a server problem yesterday?

mcm001, 2024-02-24:
Oh nice! Maybe something weird and transient yeah. Feel free to reopen if necessary

closed 2024-02-24 by mcm001