PhotonVision/photonvision #1256: Cannot connect to dash on Romi (although photonvision running)

builder173, 2024-02-24:
I followed the "Romi Installation" guide [here](https://docs.photonvision.org/en/latest/docs/installation/sw_install/romi.html), seemingly with no issues. I'm able to connect to the wpilib dash on 10.0.0.2 but get connection refused error when trying final step from guide to open dashboard:
_After it reboots, you should be able to locate the PhotonVision UI at: http://10.0.0.2:5800/._

If I ssh into the pi I can even see that the photonvision.jar program is running. However netstat doesn't show the program binding any ports. I think I saw in codebase that logs stored in /config but not sure which directory. I'm not sure if this has always been the case, but [this](https://docs.wpilib.org/en/stable/docs/romi-robot/imaging-romi.html) WPILib page says the Romi image boots up Read-only and I noticed that PhotonVision needs write access? Could that be a new incompatibility?

```
pi@wpilibpi(rw):~$ more runCamera 
#!/bin/sh
### TYPE: builtin
echo "Waiting 5 seconds..."
sleep 10000
#exec /usr/local/frc/bin/multiCameraServer
pi@wpilibpi(rw):~$ ps -ef|grep photon
root         313       1  0 12:23 ?        00:00:07 /usr/bin/java -Xmx512m -jar /opt/photonvision/photonvision.jar
pi          1332     721  0 12:45 pts/0    00:00:00 grep --color=auto photon
pi@wpilibpi(rw):~$ netstat | grep 5800
```

**Platform:**
 - Hardware Platform : raspberry pi 3B+
 - Network Configuration : romi serves as wireless access point
 - PhotonVision Version: ran install.sh which should be latest
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: Logitech C270

mcm001, 2024-02-24:
Photon needs a writable FS to extract native libraries yeah. Try journalctl -u photonvision.service?

builder173, 2024-02-25:
@mcm001 - thanks, that does show the log. It's limited to whatever appeared after FS is made writable.

It works to make the FS writable in the WPILib dash and then restart the service. After this I was able to open the dashboard. 
```
>sudo systemctl stop photonvision
>sudo systemctl start photonvision
```

closed 2024-02-25 by builder173