PhotonVision/photonvision #1259: Speed OrangePi boot time

fovea1959, 2024-02-27:
**Is your feature request related to a problem? Please describe.**
Trying to get vision booted as quickly as possible.

**Describe the solution you'd like**
Disable unneeded services that are shown by systemd-analyze to slow boot.
Disable network manager if static address set.

**Describe alternatives you've considered**
A short list:
* Disable the Bluetooth service. `systemctl disable ap6275p-bluetooth`
* Disable the cloud-init service. `touch /etc/cloud/cloud-init.disabled`
* Somehow get /etc/network/interfaces set correctly, and NetworkManager service disabled if a static address was requested.

...and changing the /etc/passwd entry for userid 'pi' to use bash instead of sh would be nice :)

mcm001, 2024-02-27:
Big fan of stuff booting faster yeah. 

Wait, do we not kill Bluetooth and WiFi in our image yet? Yikes. We need to do that

Juniormunk, 2024-10-09:
fixed in [#19](https://github.com/PhotonVision/photon-image-modifier/pull/19)

closed 2024-10-09 by Juniormunk