PhotonVision/photonvision #1261: Static up can’t be changed on orange pi 5

brhea, 2024-03-03:
When trying to edit the static up address on an orange pi 5 the address is editable and will save but the actual address is not changed. 

The pi insists on using 10.TE.AM.92
We try to edit to 10.TE.AM.93 and address will not change.

This is not an urgent issue as we can just use the .92 address but it  initially took us a while to discover the static ip address being used. 

Running latest version of photonvision.

gartaud, 2024-03-11:
See post 14 at https://www.chiefdelphi.com/t/photonvision-ojectdetection-camera-sometimes-missing-on-boot/458083/14

The UI is not intuitive, but that post might help. 

For the PhotonVision team, I don't think that it makes sense to go under "Advanced networking" to be able to change the IP address. You probably should rethink how elements are arranged on that page.

brhea, 2024-03-14:
I clicked on the drop down and selected the only interface listed and saved. The IP address was not changed.

crschardt, 2024-03-15:
Please post you logs and a screen shot of photonvision showing how you have configured the network.

brhea, 2024-03-16:
See attached screenshot of network config:

[image]

crschardt, 2024-03-16:
I meant "download current log" and post the entire text file.

brhea, 2024-03-19:
Find attached uploaded log:
I plan to upgrade to version 2024.3.1 today. Perhaps this will fix the issue. There seemed to have been a few network changes in this release.

[photonvision-journalctl (1).txt](https://github.com/PhotonVision/photonvision/files/14655050/photonvision-journalctl.1.txt)

Juniormunk, 2024-10-28:
Fixed by #1478

closed 2024-10-28 by Juniormunk