PhotonVision/photonvision #1271: NT Heartbeat continues even when camera is not active

willtoth, 2024-03-11:
I can see the camera in the UI, and the heartbeat in NT is going, but no data is published, and no image comes through. The only way to detect that this condition is occurring from what I can tell is to look at journalctl, and only way to recover is manually unplug/replug or restart the bus.

This happened right after boot. It's not clear how to reproduce, perhaps this is some weird race condition due to fast boot being enabled?

I did not get a chance to view dmesg before recovering.

[image]

**Platform:**
 - Linux x86 - Clear Linux with kernel 6.7.2-1406
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Network switch
 - PhotonVision Version: 2024.2.5
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Brave
 - Camera(s) Used: OV2311 (x2)

[Message on discord](https://discord.com/channels/725836368059826228/725848198794706994/1216119828469907598)

gartaud, 2024-03-11:
Are the symptoms that you are observing that you can see a feed in the built-in UI but cannot see them in a remote client such as the Shuffleboard as described in post 7 of https://www.chiefdelphi.com/t/photonvision-ojectdetection-camera-sometimes-missing-on-boot/458083/7 ?

willtoth, 2024-03-11:
I did not try getting a feed using the NT stream, but I did not see a stream in the web UI.

gartaud, 2024-03-11:
Thanks! Is it a bit like the one described in post 5 then?

https://www.chiefdelphi.com/t/photonvision-ojectdetection-camera-sometimes-missing-on-boot/458083/5

willtoth, 2024-03-11:
Its a bit hard to tell without the journalctrl output, but I also don't think its the same as this one. I did not notice any streams interfering with others. Just that the camera page did not have _any_ stream for the one I cared about, it was clearly detected, as the logs show an error message, but PV keeps the heartbeat going as if everything is fine.

mcm001, 2024-03-11:
https://github.com/PhotonVision/photonvision/blob/d8f82bf9ee98e9bdcfbde529ce9ded5c5b7ff964/photon-core/src/main/java/org/photonvision/vision/frame/provider/USBFrameProvider.java#L43
https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/cscore/CvSink.html#grabFrame(org.opencv.core.Mat,double)
We don't check error codes here, which means yeah we send a 0x0 mat through the rest of the code. Which happily processes it (doing no work, as it's empty) and sends the targeting output code zero targets but a happy I'm alive and producing results. We should probably Not do that

closed 2024-03-22 by mcm001