PhotonVision/photonvision #1274: Add option to create SimCameraProperties from exported calibration JSON

srimanachanta, 2024-03-12:
**Is your feature request related to a problem? Please describe.**
There should be an API option to create a `SimCameraProperties` object from an exported JSON file. If not the whole thing, as much as possible.

**Describe the solution you'd like**
JSON parser for loading the file and then setting the properties, Jackson is already included?

**Describe alternatives you've considered**
Manually adding coefficients to constants.

**Additional context**
This would be for SIM only so it makes sense to allow people to quickly change settings.

srimanachanta, 2024-03-12:
NVM this exists, im done

closed 2024-03-12 by srimanachanta

mcm001, 2024-03-12:
> NVM this exists, im dumb

It does? Would be good to get an example in the docs/examples if not

Best,
Matthew Morley
BSME candidate, Northeastern University
(408) 368-2815