PhotonVision/photonvision #1279: Limelight 3G Support

dmfgold, 2024-03-16:
**Is your feature request related to a problem? Please describe.**
- With the release of the Limelight 3G, support for PhotonVision with the introduction of a Limelight3G image to avoid conflicts would be optimal

**Describe the solution you'd like**
- The release of a new PhotonVision image for the Limelight 3G

**Describe alternatives you've considered**
- Using the Limelight 3 image, although this can cause potential problems.

mcm001, 2024-03-16:
If you want a LL3g image, given none of us contributors have one in hand, the fastest way to get support is to PR it :)))

mcm001, 2024-03-16:
I suspect replacing the ov5647 line with dtoverlay=ov9281 might work

BytingBulldogs3539, 2024-03-24:
I created a PR to add the ll3g however I don't have the hardware to test it but you could test it. https://github.com/PhotonVision/photon-image-modifier/pull/14

studentkrisk, 2024-04-05:
I tried changing the dtoverlay line and it worked with my 3g, but the boot times are long and it sometimes doesn't boot. any ideas?

mcm001, 2024-04-05:
Do any huge waits stand out to you in dmesg logs?

Sam948-byte, 2025-02-09:
Close per #1727

closed 2025-02-09 by mcm001