PhotonVision/photonvision #1285: Cameras start in Driver Mode on boot on OrangePi 5.

crschardt, 2024-03-17:
**Describe the bug**
Frequently, all of the cameras will be set to DriverMode when photonvision starts. This has been observed on the OrangePi 5.

**To Reproduce**
1. Install the 2024.2.10 on and OrangePi 5
2. Start photonvision and configure a camera with a pipeline
3. Shut down the OrangePi 5 by pulling the power
4. Restart and check the camera pipeline
5. You may have to repeat a few times to trigger this bug 

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): OrangePi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): Attached to home router, but also reported happening on robot connected to the radio through a switch
 - PhotonVision Version: 2024.2.10
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Edge
 - Camera(s) Used: Arducam OV9281
 
** Root Cause Investigation:**
I ran the VSCode remote debugger on photonvision and found that there are two threads that are activated when the pipeline changes. To the best of my understanding, here's what is happening:
1) VisionRunner Thread:
  - VisionRunner.update() is continuously polling the current pipeline via  a call to PipelineManager.getCurrentPipeline()
  - This calls PipelineManager.updatePipelineFromRequested()
  - If the pipeline index has changed, updatePipelineFromRequested() will update the value of currentPipeline
2) DataChangeEventDispatch Thread:
  - DataChangeService dispatches a "changePipeline" event from the queue
  - This triggers the VisionModuleChangeSupplier to call VisionModule.saveAndBroadcastAll()
  - This eventually leads to ConfigManager saving the CameraConfig

If the CameraConfig is saved before the VisionRunner thread is able to update the pipeline, then it saves the old values rather than the updated value for the pipeline. It also causes the database to be out of sync with the current pipeline and UI.

srimanachanta, 2024-03-17:
Ditto on this problem

mcm001, 2024-03-17:
Does this fix it? https://github.com/PhotonVision/photonvision/pull/1286

crschardt, 2024-03-17:
It looks promising. I've rebooted several times without any pipeline problems. I'll continue to test.

mcm001, 2024-03-21:
Closing as addressed

closed 2024-03-21 by mcm001