PhotonVision/photonvision #1308: Use camera to position AprilTags to the right location.

Schroedes, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
Making sure you position the AprilTags around a field is very difficult to get exact, by building a feature to assist in placing these tags, it would alleviate most doubt when tracking down pose estimation discrepancies. 
**Describe the solution you'd like**

A clear and concise description of what you want to happen.
A feature built into PhotonVision where you place a camera at a known pose, and upload a mapping of where the AprilTags are supposed to be placed. Based on this predetermined camera pose we can determine where in the camera's frame the tag should be positioned(Potentially a green box displayed on where the tag should be placed and then indicators when the translation and rotation are correct). 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Roman-Tech-Plus, 2024-04-09:
I feel like relying on robot placement is somewhat of a recipe for disaster. Potentially, a more SLAM esque approach like 971 would be preferable as it simply bases the poses of all other tags off a single known good tag. That said, after the field tolerance issues at early seosn events, GDC seems to have gotten thing more together by week 5. Both us and 971 found no significant placement abnormalities with our field solvers. That said, this could definitely be a useful feature for practice fields or field tolerance edge cases.

Gold856, 2025-04-14:
Out of scope for PV.

closed 2025-04-14 by Gold856