PhotonVision/photonvision #1315: Support for Orange Pi 5 Pro

grahamgraham, 2024-05-02:
**Is your feature request related to a problem? Please describe.**
None of the current images work with the newly release Orange Pi 5 Pro.
I got my hands on an OPI 5 Pro, and neither the 5 nor the 5 plus images will work (it does not boot). The linux and ubuntu images on OPI's website for the 5 Pro do boot. 

**Describe the solution you'd like**
A release of Photon Vision for the Orange Pi 5 Pro

**Describe alternatives you've considered**
I've considered (and will attempt) using the "Other Debian-Based Co-Processor Install", but boot times with OPI's default image aren't super conducive for FIRST competitions.

srimanachanta, 2024-05-02:
Take a look at this instead of the default image: https://www.armbian.com/orangepi-5/

grahamgraham, 2024-05-02:
Armbian allows me to boot into it, but there's other issues (unable to detect ethernet interface).

Does that mean if Armbian works, the current release of PV for OPI 5 should work?

srimanachanta, 2024-05-02:
Armbian's image is significantly faster for booting than Orange Pi's original image which would solve that issue. You would then need to install PV on that following the docs you sent. This would be temporary until a first class image could be made. Based on Armbian's build, a device-tree might be present regardless of board type in the image so we might be able to adapt our current image-gen with that in mind

grahamgraham, 2024-05-02:
Ok, that makes sense. 
There's still ethernet issues with the Armbian images out there, but I am happy to report that the OPI 5 Pro bookworm server build that's provided on OPI's site is working well with PhotonVision. 

Please let me know if you need a tester for the OPI 5 Pro's first class image.

srimanachanta, 2024-05-02:
Upon further investigation, the upstream image provider actually does have an image for OPi5-Pro found [here](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.0.0/ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-pro.img.xz). This is the image we use for all our image builds. Run [this](https://github.com/PhotonVision/photon-image-modifier/blob/master/install_opi5.sh) script to actually install and setup PV but know that this hasn't been tested on the OPi5 Pro by the PV dev team.

grahamgraham, 2024-05-02:
Out of curiosity, I see that that image is a desktop image. When PV releases an approved version, is the build typically CLI or desktop?

srimanachanta, 2024-05-02:
> Out of curiosity, I see that that image is a desktop image. When PV releases an approved version, is the build typically CLI or desktop?

Server, so command line only.

crschardt, 2024-05-04:
With 24.04 you may also have to install gcc or else photonvision will crash at startup.

roberttuttle, 2024-05-08:
We tried upstream 24.04 (2.0.0) server image linked above for a new Orange Pi 5 Pro we received and it failed to boot (kernel panic) ... so it doesn't appear stable yet.

In an alternate path for the near-term, we went down the path of using the official Ubuntu 24.04 / Jammy image from the Orange Pi downloads and then ran through the PV image install steps. PV was worked correctly with our OV9281 cams at that point, BUT, object detection pipeline (RKNN) was not offered as an option in the pipeline list.

I researched that Ubuntu/OPi5 image and found that all the required libs for RKNN were in place and working outside PV.

I then dug into the PV code and logs and found that the object detection feature being offered was dependent on the platform detection successfully parsing the model name from the device tree with "Orange Pi 5". Unfortunately, the Orange Pi 5 Pro image has a compiled device tree model name of "RK3588S Opi 5 Pro" which of course doesn't match the string comparison in the PV Java code here: https://github.com/PhotonVision/photonvision/blob/00c2a257305c421231d150587ffb7c65fb963742/photon-core/src/main/java/org/photonvision/common/hardware/Platform.java#L217

As a local fix without offering a PV code fix via PR, I was able to edit the device tree model name using the native device-tree-compiler tool with the command: `sudo fdtput -t s /boot/dtb/rockchip/rk3588s-orangepi-5-pro.dtb / model "RK3588S Orange Pi 5 Pro"` (followed by a reboot)

After reboot, PV correctly matched the platform and allowed object detection pipeline to show up (and work successfully with Crescendo notes out of the box).

Obviously, longer term fix is to update PV code to support whatever official Orange PI 5 Pro device tree model name is supplied for the board/images.

grahamgraham, 2024-05-08:
> We tried upstream 24.04 (2.0.0) server image linked above for a new Orange Pi 5 Pro we received and it failed to boot (kernel panic) ... so it doesn't appear stable yet.

Funny enough, I was able to get that server image to boot without issue. However, PV wouldn't install properly. 
I ended up using the OPi 5 Pro server release that's on OPi's site. For offseason it will work well enough for testing. 

I'm stoked because this is a much nicer solution than our Limelight 3, and way cheaper than the LL3G.

srimanachanta, 2024-05-08:
Happy to hear that worked, you might want to try installing dietpi onto orange pi's image in order to clean it up, I'm not sure how much it will reduce boot times though.

grahamgraham, 2024-05-08:
Ah, that's a great idea. I'll give that a shot!

crschardt, 2024-07-30:
There is a new image available which supports the OrangePi pro. Please download the correct image from https://github.com/PhotonVision/photon-image-modifier/releases/tag/Dev and let me know if it works.

grahamgraham, 2024-07-31:
> There is a new image available which supports the OrangePi5 Pro. Please download the correct image from https://github.com/PhotonVision/photon-image-modifier/releases/tag/Dev and let me know if it works.

Appears to be working so far. 
[image]
Version is reported as v2024.3.1, which I do not see as previous version, so I assume that's the Dev version?

If that's the case, I will be working on getting this installed to the eMMC

crschardt, 2024-08-01:
@grahamgraham - thank you for testing. 2024.3.1 is the current release of Photonvision, not to be confused with the development image version, which is

mcm001, 2024-08-01:
The fact that your pi boots at all is sufficient - the JAR is swapped out downstream of this.

grahamgraham, 2024-08-01:
> The fact that your pi boots at all is sufficient - the JAR is swapped out downstream of this.

Good deal. With that said, I've also installed it on the eMMC and have been running it using 2 cameras without issues. Thanks for everybody's help on this!

crschardt, 2024-08-25:
Good to hear. Since we have confirmation from a couple people that it's working, I'm going to close this. Please open a new issue if you come across any other problems.

closed 2024-08-25 by crschardt