PhotonVision/photonvision #1318: Ignore Tags

lakshag32, 2024-05-12:
It would be useful to both our team(and likely others) if we had the ability to ignore certain tags, both in single tag mode and multi-tag mode. If we could ignore tags, we could ignore tags that move around a lot on the field due to robot collisions, improving odometry.

mcm001, 2024-05-13:
You can already do this, in a roundabout way, by removing tags from the apriltag field layout used or by doing extra processing on-rio to filter single-tag results. Is that enough, and we just need to document it is an option?

rwlee, 2024-09-24:
@mcm001  -- I think the confusion here is because people expect to be able to reconfigure it in code when there are actually 2 configuration points (rio side and coprocessor side). If you change the `AprilTagFieldLayout` in code, you will have a inconsistency between the rio and coprocessor.

https://docs.photonvision.org/en/latest/docs/apriltag-pipelines/multitag.html#updating-the-field-layout

rwlee, 2024-09-24:
Also, this seems to effectively be a duplicate issue of #729

mcm001, 2024-09-24:
Yeah agreed - let's move conversation there

closed 2024-09-24 by mcm001