PhotonVision/photonvision #1319: Photonvision settings not being saved.

lakshag32, 2024-05-12:
**Describe the bug**
Photonvision settings are sometimes reset after a UI restart/coprocessor shutdown and reboot. Can't recall exactly which one.

**Setup**: 
1. Hardware: Orange Pi 5
2. Network config: Network switch
3. Photonvision version: 2024.2.9
4. Browser: Chrome, unsure of version
5. Camera: Arducam USB OV9281

**To Reproduce:**
Steps to reproduce the behavior:
1. Adjust photonvision settings
2. Restart photonvision/reboot orange pi
3. Observe that the settings we changed have been reset

srimanachanta, 2024-05-22:
Please include logs for this as the UI should error/response when settings weren't properly saved. Will need the logs to trace this down further.

crschardt, 2024-06-11:
Any more info on this issue? If not, I'm going to close it.

lakshag32, 2024-06-11:
Apologies for the late reply. Here are some logs that may shed some light on the issue. Let me know if they help. 
[photonvision-journalctl_6_5_24.txt](https://github.com/user-attachments/files/15793514/photonvision-journalctl_6_5_24.txt)

crschardt, 2024-06-12:
I see you're still using 2024.2.9, however the newest stable version is [2024.3.1](https://github.com/PhotonVision/photonvision/releases/tag/v2024.3.1), which has a lot of improvements. The bug you're encountering could be already fixed. Please upgrade to this version and let me know if you are still having the problem.

lakshag32, 2024-06-12:
I see, thank you!

mcm001, 2024-06-12:
The diff is here: https://github.com/PhotonVision/photonvision/compare/v2024.2.9...v2024.3.1 . Nothing here jumps out to me as a smoking gun, though

crschardt, 2024-08-04:
@lakshag32 - are you still experiencing this bug?

Juniormunk, 2024-10-28:
We need to revisit this. We changed some things around that I believe may have caused this issue.

samfreund, 2025-04-12:
Can we close this? I haven't been seeing reports of it this season.

mcm001, 2025-04-12:
Please re-open if anyone sees this again!!

closed 2025-04-12 by mcm001