PhotonVision/photonvision #1324: PhotonVision does not Run on Arm32

scarmain, 2024-05-18:
**Describe the bug**
PhotonVision does not Run on Arm32.  It looks like 64bit JNIs are used in the 32 bit build.

**To Reproduce**
I created my own Pi image to run on a Pi Zero W.  (using the RPi Os Lite 32-bit 'bookworm' version to save memory on the 512MB device)  When running PhotonVision, I got these errors on startup:

pi@raspberrypi:~ $ java -jar photonvision-v2024.3.1-linuxarm32.jar 
[2024-05-19 00:30:34] [General - Main] [INFO] Native libraries loaded.
OpenJDK Client VM warning: You have loaded library /tmp/libphotonlibcamera.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
[2024-05-19 00:30:34] [Camera - LibCameraJNILoader] [ERROR] Couldn't load shared object photonlibcamera: /tmp/libphotonlibcamera.so: /tmp/libphotonlibcamera.so: wrong ELF class: ELFCLASS64 (Possible cause: can't load AARCH64 .so on a ARM platform)
java.lang.UnsatisfiedLinkError: /tmp/libphotonlibcamera.so: /tmp/libphotonlibcamera.so: wrong ELF class: ELFCLASS64 (Possible cause: can't load AARCH64 .so on a ARM platform)
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2394)
	at java.base/java.lang.Runtime.load0(Runtime.java:755)
	at java.base/java.lang.System.load(System.java:1957)
	at org.photonvision.raspi.LibCameraJNILoader.forceLoad(LibCameraJNILoader.java:64)
	at org.photonvision.Main.main(Main.java:365)
[2024-05-19 00:30:34] [General - Main] [ERROR] Platform does not support RKNN based machine learning!
[2024-05-19 00:30:34] [General - Main] [WARN] Failed to load mrcal-JNI! Camera calibration will fall back to opencv
Unable to load mrcal JNI!
[2024-05-19 00:30:34] [General - Main] [INFO] Logging initialized in debug mode.
[2024-05-19 00:30:34] [General - ShellExec] [DEBUG] Executing "cat /proc/device-tree/model"
[2024-05-19 00:30:34] [General - ShellExec] [DEBUG] Got exit code 0
[2024-05-19 00:30:34] [General - Main] [INFO] Starting PhotonVision version v2024.3.1 on Linux Raspbian 32-bit (Pi ZERO_2_W)
...

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Raspberry Pi Zero 2 W, RPi Os Lite 32-bit version 'bookworm' arm32 build
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): On-Board Wifi
 - PhotonVision Version: v2024.3.1
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): (not relevant, used SSH to start it, but use FireFox)
 - Camera(s) Used: Camera Port with Pi Cam v1.3

**Additional context**
Add any other context about the problem here.

Bankst, 2024-05-19:
Lack of ARM32 support is not a bug

closed 2024-05-19 by Bankst

scarmain, 2024-05-19:
Why is this not a supported platform?  This is one of the released architectures of PhotonVision.  If it not supported, you should probably stop building it...

[image]

mcm001, 2024-05-19:
For a little more context here: none of our contributors own arm32 SBCs to test on, and it sounds like nobody has tried it since the start of 2024 either. Congrats on being the first (and last)! 

If you want to, it is probably technically possible to update our libcamera and mrcal libraries to build for arm32 in the future -- I would consider those PRs anyways -- but all the common frc platforms are arm64 anyway hence the lack of testing.