PhotonVision/photonvision #1331: Vendor Library installation issue

LarryMoe11, 2024-05-29:
**Describe the bug**
When installing the vendor library, I believe it is being grabbed from the wrong source. When I got the correct vendor library link (https://maven.photonvision.org/releases/org/photonvision/photonlib-json/1.0/photonlib-json-1.0.json), it still tried downloading the java dependecies from the old repository, which no longer works

**To Reproduce**
Steps to reproduce the behavior:
1. Copy the installation link from the documentation at: https://docs.photonvision.org/en/latest/docs/programming/photonlib/adding-vendordep.html
2. Install the link copied using "Manage Vendor Libraries" -> "Install new libraries (online)"

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Windows x64
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): N/A
 - PhotonVision Version: N/A
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Opera GX
 - Camera(s) Used: N/A

**Additional context**
Output from console after installing: 
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Dev\Robotics\2024-Libs\build.gradle' line: 87

* What went wrong:
Could not determine the dependencies of task ':jar'.
> Could not resolve all files for configuration ':runtimeClasspath'.
   > Could not find org.photonvision:photonlib-java:v2024.3.1.
     Searched in the following locations:
       - file:/C:/Users/Public/wpilib/2024/maven/org/photonvision/photonlib-java/v2024.3.1/photonlib-java-v2024.3.1.pom
       - https://frcmaven.wpi.edu/artifactory/vendor-mvn-release/org/photonvision/photonlib-java/v2024.3.1/photonlib-java-v2024.3.1.pom
       - https://maven.photonvision.org/org/photonvision/photonlib-java/v2024.3.1/photonlib-java-v2024.3.1.pom
       - https://repo.maven.apache.org/maven2/org/photonvision/photonlib-java/v2024.3.1/photonlib-java-v2024.3.1.pom
     Required by:
         project :
   > Could not find org.photonvision:photontargeting-java:v2024.3.1.
     Searched in the following locations:
       - file:/C:/Users/Public/wpilib/2024/maven/org/photonvision/photontargeting-java/v2024.3.1/photontargeting-java-v2024.3.1.pom 
       - https://frcmaven.wpi.edu/artifactory/vendor-mvn-release/org/photonvision/photontargeting-java/v2024.3.1/photontargeting-java-v2024.3.1.pom
       - https://maven.photonvision.org/org/photonvision/photontargeting-java/v2024.3.1/photontargeting-java-v2024.3.1.pom
       - https://repo.maven.apache.org/maven2/org/photonvision/photontargeting-java/v2024.3.1/photontargeting-java-v2024.3.1.pom    
     Required by:
         project :

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 47s

mcm001, 2024-05-29:
Hi! I was in the middle of transfering our Maven servers over to Reposilite and setting up redirects. I just tested with a blank robot project, and it seems to be working -- can you confirm?

LarryMoe11, 2024-05-29:
That worked, thank you!

mcm001, 2024-05-29:
For future reference, only v2024.3.1 has been transferred over at this point. Other versions can be backported if we need them.

closed 2024-05-29 by mcm001