PhotonVision/photonvision #1338: Changing resolutions on the Aruco pipeline causes PhotonVision to crash.

cb144p, 2024-06-04:
**The issue**
When changing the resolution of the camera to 1600x1200 in the Aruco pipeline PhotonVision crashes. Some resolution changes were completed, but not always. The UI is unable to contact the coprocessor and trying to reconnect to PhotonVision on another device also fails. The non response does not persist through power cycles.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the micro sd card with orange pi image version 2024.2.9
2. Connect the camera to the device.
3. Connect power to Orange Pi.
4. Select the camera from the dropdown.
5. Change the pipeline to Aruco.
6. Change the resolution to 1600x1200@50fps.
7. Watch it hang.

**Logs**
[Log before reflash](https://github.com/user-attachments/files/15568164/photonvision-journalctl_1.txt).
[Log after clean image](https://github.com/user-attachments/files/15568172/photonvision-journalctl_3.txt).

**Platform:**
 - Orange Pi 5
 - With and without network switch
 - PhotonVision Version 2024.2.9
 - Chrome and Safari
 - [Arducam OV2311](https://www.arducam.com/product/arducam-2mp-global-shutter-usb-camera-board-for-computer-50fps-ov2311-monochrome-uvc-webcam-module-with-low-distortion-m12-lens-without-microphones-compatible-with-windows-linux-android-and-mac-os/)

**Additional context**
PhotonVision would also crash when trying to calibrate the camera. It wasn't an instant crash triggered by something immediate like the resolution change in the Aruco pipeline, PhotonVision would just hang randomly throughout the process. The calibration process was also generally stuttery, with the program being non responsive temporarily and then running smoothly for a few seconds. We were using auto exposure because the camera wasn't responding to the manual exposure adjustments properly. PhotonVision seemed to just be generally unstable with changes related to the camera. The changes that were successful often had long delays between the action requested and the action completing.

Juniormunk, 2024-10-28:
We need someone to test and see if this bug was fixed I some of the latest updates

mcm001, 2025-03-20:
OBE, reopen if still an issue

closed 2025-03-20 by mcm001