PhotonVision/photonvision #1339: Support GPU Accelerated FrameCapture on all OpenGL supported platforms.

srimanachanta, 2024-06-05:
Currently, we use libcamera + OpenGL for GPU acceleration but we limit it to Raspberry Pi only. OpenGL is supported by ARM Mali GPUs as well so we should be able to adapt the current driver to support both VideoCore and Mali GPUs by leveraging OpenGL. If the issue lies with libcamera, we could revert to V4l2.

mcm001, 2024-06-05:
We have a prototype using JOGL for HSV, but without zero-copy, it was slower than on-CPU.

ramalamadingdong, 2024-06-05:
can I see this prototype? I am doing benchmarks for a GPU based robotics kit and would like to run those benchmarks using Photonvision. 
You can make v4l usb camera into shared memory using something like this [V4L-RS](https://github.com/raymanfx/libv4l-rs) 

Or maybe gstreamer. 

I am really happy to help do some / most of the dev work required to get this going but would like to get plugged in to current efforts first. 

Last question:
If I use libcamera why is it limited to RPI?