PhotonVision/photonvision #1353: Orange PI 5 Pro Failed Install

ZenoCoding, 2024-06-27:
**Describe the bug**
I am unable to install PhotonVision on my Orange Pi 5 Pro. After using an image provided by the "upstream" source of the images PhotonVision uses and installing gcc, PhotonVision reports that it is installed successfully but does not connect when accessed through my router via direct IP.

**To Reproduce**
Steps to reproduce the behavior:
1. get latest version of ubuntu for orange pi 5 pro from: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/
2. flash orange pi it with balenaetcher onto the microsd
3. boot orange pi and connect it to the router via direct ethernet connection
4. find ip address ssh with user and password ubuntu
5. verify internet connection by sudo ping -c 4 8.8.8.8
6. update package list and install gcc with build-essential
7. run install instructions https://docs.photonvision.org/en/latest/docs/installation/sw_install/other-coprocessors.html
8. attempt to connect to photonvision via IP:5800

**Screenshots / Videos**
Install Logs (ssh): https://pastebin.com/gvynNg87
Running Logs: 
[logpvboot.txt](https://github.com/user-attachments/files/16006687/logpvboot.txt)
i spent a long time figuring out how to get this log file... i hope its the right one

i also changed my networking setup (my monitor is far away from my router and not easy to move) to the second port of the robot radio for the running logs

**Platform:**
 - Hardware Platform: Orange Pi 5 Pro
 - Network Configuration: Coprocessor --(ethernet)-> xfinity home router --(LAN)-> personal computer
 - PhotonVision Version: latest
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Arc 1.49.0 (51177)-RC
 - Camera(s) Used: Arducam OV2311

**Additional context**
The image used was the latest version of the upstream image provider that photonvision uses as described by this [comment](https://github.com/PhotonVision/photonvision/issues/1315#issuecomment-2091194677) in a similar issue.

They mention some issues that I believe have since then been patched. Ubuntu is definitely able to start, and I had thought that photonvision had been successfully installed because of this message:

```
Created PhotonVision systemd service.
PhotonVision installation successful.
```

Clearly, this is not the case, or there is something wrong with my networking configuration because I cannot connect to the webdashboard. 

A orange pi 5 pro image provided by photonvision would also be nice! Though maybe i should read the docs of the software I'm going to use before I buy things...

Any help is appreciated!

ZenoCoding, 2024-06-27:
Here's the photonvision journal log:
[pvjournal.txt](https://github.com/user-attachments/files/16007350/pvjournal.txt)

crschardt, 2024-07-30:
There is a new image available which supports the OrangePi pro. Please download the correct image from https://github.com/PhotonVision/photon-image-modifier/releases/tag/Dev and let me know if it works.

victkk, 2024-08-10:
> There is a new image available which supports the OrangePi5 Pro. Please download the correct image from https://github.com/PhotonVision/photon-image-modifier/releases/tag/Dev and let me know if it works.

I tested on my orange pi 5 pro. It works fine. (uploading cam calib file, apriltag 2D/3D pipeline) But I didn't know how to sync my photonLib version to the dev version so that I can do pose estimation on roborio. Using the v2024.3.1 will report error.
[image]

 I see the photonLib package that came with the dev release. But when I refer to the PhotonLib docs for offline installation guide( https://docs.photonvision.org/en/latest/docs/programming/photonlib/adding-vendordep.html#offline-install-java-c ). The docs says"This installation option is currently a work-in-progress. For now, we recommend users use the online installation method."

So how can I update photonLib to the dev Version? Or is there any way to bypass the photonlib version check?

mcm001, 2024-08-10:
Just change the version string in your vendor dep JSON, or https://javadocs.photonvision.org/org/photonvision/PhotonCamera.html#setVersionCheckEnabled(boolean) . We need to document this so feel free to open a issue or make a PR to get this updated :)))

victkk, 2024-08-11:
> Just change the version string in your vendor dep JSON, or https://javadocs.photonvision.org/org/photonvision/PhotonCamera.html#setVersionCheckEnabled(boolean) . We need to document this so feel free to open a issue or make a PR to get this updated :)))

Ok, I will make a PR in a couple days.

victkk, 2024-08-13:
"change the version string in your vendor dep JSON" worked with a bit of extra work. I had to adjust my code because the constructor for PhotonPoseEstimator changed in PhotonLib dev-v2024.3.0-49-g67463a02（I originally used v2024.3.1)

"setVersionCheckEnabled(False)" failed though it does prevent the program from crashing. It's probably because the rawdata format changed a lot across the two version and parsing the rawdata from a dev version coprocessor with the v2024.3.1 photonLib cause misalignment. For example it would parse the two bytes that stands for heartbeat as latency.

victkk, 2024-08-13:
> Just change the version string in your vendor dep JSON, or https://javadocs.photonvision.org/org/photonvision/PhotonCamera.html#setVersionCheckEnabled(boolean) . We need to document this so feel free to open a issue or make a PR to get this updated :)))

"change the version string in your vendor dep JSON" worked with a bit of extra work. I had to adjust my code because the constructor for PhotonPoseEstimator changed in PhotonLib dev-v2024.3.0-49-g67463a02（I originally used v2024.3.1)

"setVersionCheckEnabled(False)" failed though it does prevent the program from crashing. It's probably because the rawdata format changed a lot across the two version and parsing the rawdata from a dev version coprocessor with the v2024.3.1 photonLib cause misalignment. For example it would parse the two bytes that stands for heartbeat as latency.

crschardt, 2024-09-19:
The OrangePi5pro image is available now. I'm closing this issue. Please open a new issue if you have trouble with the image.

closed 2024-09-19 by crschardt