PhotonVision/photonvision #1354: Support uploading and selection of custom user models

Alextopher, 2024-06-28:
**Is your feature request related to a problem? Please describe.**
The existing workflow for using custom object detection models isn't well suited to the task of experimenting with custom models or sharing resources across teams. Being required to sign into the coprocessor is an unnecessary barrier for team's who are just interested in downloading & using someone else's model.

**Describe the solution you'd like**
Add support for uploading custom models and a way to switch between models.

**Describe alternatives you've considered**
Module selection alone would be a good start. Teams could continue to use `scp` to upload and manage models but then have access to an interface to pick-and-choose the model of interest.

**Additional context**

mcm001, 2024-06-28:
I think the simplest, MVP implementation of this feature is a CRUD interface in our settings tab to upload arbitrary files, and a dropdown in neural network pipelines to select what file name I want to use. This covers 90% of what users actually need I think.

Alextopher, 2024-06-28:
Sounds good! Tomorrow I should have the time to gauge if I'm able to make that contribution.

Alextopher, 2024-06-30:
I'll take this on - but no promises on the timeline.

mcm001, 2024-06-30:
Should (?) be fixed! Note my comment for windows users needing to use the FRCK JDK

Sam948-byte, 2025-02-09:
Close as completed.

closed 2025-03-20 by spacey-sooty