PhotonVision/photonvision #1380: Document SD Card choice

Juniormunk, 2024-07-30:
Industrial SD cards like this [one](https://www.amazon.com/SanDisk-Industrial-MicroSD-UHS-I-SDSDQAF3-016G/dp/B07CV344WJ/) provide a lot of preventative measures such as wide voltage and temperature range. They are also "waterproof, shock and vibration proof, X-ray proof, magnet proof, impact proof". While some of those are just nice to have the one that seems quite important is "Advanced memory management FW features power immunity, auto/manual read refresh, ECC, wear leveling". This should help prevent corruption when you power off while writing to the SD card. We could only trace our random shutdowns back to the SD card. We will only be using industrial cards now.

mcm001, 2024-07-30:
SD card quality mattering I think was also discovered with the Rio 2 -- does wpilib or NI have any recommendations there we can borrow from?

mcm001, 2024-10-26:
Bump here - we should absolutely do this

Sam948-byte, 2025-01-08:
This looks like it was finished, and can probably be closed.

closed 2025-01-08 by Juniormunk