PhotonVision/photonvision #1386: Make Client Side Websocket Requests Asynchronous

srimanachanta, 2024-08-02:
Currently, websocket changes sent from the UI to the backend are just assumed to be successful, instead, the backend should return a `WebsocketReponse` struct similar to an HTTP response. By doing this, we can make sure that websocket-based changes are suspenseful.

To implement this we need to make the backend echo back a `WebsocketReponse` object back to the UI which has an async promise waiting for this message. Multiple requests can be handled by including the request handler within the response so the promise can determine the intended response. Also means that the API for this can be abstracted well for both the frontend and backend.

mcm001, 2024-08-02:
Mom can we have HTTP requests?

No, we have HTTP requests at home

HTTP at home:

Alextopher, 2024-08-04:
I wonder if you could go into a bit more detail about what kind of improvements we'd see from this. Will this somehow communicate errors or just communicate the network delay to the user?