PhotonVision/photonvision #1387: Disable streams

ItayNaumann, 2024-08-03:
During the season, we've used photonvision, and we had bandwidth problems. We wanted to see a certain stream, but only 1 stream at a time while stopping all of the others. Not seeing the stream didn't disable it, and it took a lot of bandwidth with it.

A way to disable each one of the streams, probably via a NT entry.

Could also be cool maybe a "focus mode" for when there are a couple of cameras and wanting to show only one of the stream at a time, maybe a NT entry that uses the port's number to choose focus

mcm001, 2024-08-04:
If not viewing a stream doesn't decrease bandwidth, I would argue that actually a bug and we should open an issue either way to try and triage it.

Juniormunk, 2024-08-10:
How were you viewing the streams? What app were you using?

ItayNaumann, 2024-08-10:
We were listening from the camera publisher entry in our dashboard

Juniormunk, 2024-08-10:
What resolution were the streams?

ItayNaumann, 2024-08-10:
One colorfull, one greyscale. OV 9281 and OV 9282. I think that both had a stream of 640 by 480

mcm001, 2024-08-11:
> We were listening from the camera publisher entry in our dashboard

Which dashboard software did you use? And did you also have the web ui open at the same time?

mcm001, 2024-10-13:
Is this still the case after merging https://github.com/PhotonVision/photonvision/pull/1437 ?

samfreund, 2025-04-14:
Closing as stale, reopen if it's still an issue.

closed 2025-04-14 by samfreund