PhotonVision/photonvision #1405: Simulated Pose Estimation Only Works When In Motion

falOn-Dev, 2024-08-30:
**Describe the bug**
I'm not sure if this is intended, but when running simulated PhotonVision pose estimation, it only seems to be outputting poses when the robot is in motion. When it's stationary but still sees targets, the `Optional<EstimatedRobotPose>` starts returning as an empty optional.

**To Reproduce**
Steps to reproduce the behavior:
1. [Simulate my robot code](https://github.com/falOn-Dev/KrakenSwerve/tree/akit-custom-swerve) (use the linked branch)
2. Drive around
3. View the `isPresent` field under `RealOutputs/vision`

**Screenshots / Videos**

https://github.com/user-attachments/assets/a5603eb5-6085-47d5-a4ca-6e91f443872c

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Windows x64, no co-processor
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): N/A (Sim)
 - PhotonVision Version: PhotonLib is `v2024.3.1`, no PV running on a co-processor
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): N/A
 - Camera(s) Used: N/A

**Additional context**
I'm not completely sure this is a bug with PhotonVision, however I've seen another person with a similar issue, and I was asked by Matt to submit this.

amquake, 2024-09-05:
https://github.com/PhotonVision/photonvision/blob/7271c950e10401fa97c7b9e910ef14cb316c0da0/photon-lib/src/main/java/org/photonvision/PhotonPoseEstimator.java#L135-L144

https://github.com/PhotonVision/photonvision/blob/7271c950e10401fa97c7b9e910ef14cb316c0da0/photon-lib/src/main/java/org/photonvision/PhotonPoseEstimator.java#L357-L366

This should be checking if the updated pose is different from the last, or that the timestamp is different. An empty estimate should only be sent if both of those conditions are false. One thing I'm not certain on is if the NT subscriber updates the timestamp for identical values:
https://github.com/PhotonVision/photonvision/blob/7271c950e10401fa97c7b9e910ef14cb316c0da0/photon-lib/src/main/java/org/photonvision/PhotonCamera.java#L182-L185

`getLastChange()` here possibly not updating the timestamp, which would cause the behavior you see.

Your simulated camera has no noise added to it, which is causing the estimated pose while stationary to be completely stable. This probably wouldn't happen on a real robot.

gerth2, 2024-09-05:
>One thing I'm not certain on is if the NT subscriber updates the timestamp for identical values:

With NT4 - yes - there should be one publish operation per piece of info in the camera frame. Data can be identical, but timestamp will advance.

mcm001, 2025-03-20:
Closing in favor of #1180 , as these look like duplicates?

closed 2025-03-20 by mcm001

amquake, 2026-01-20:
(duplicate)