PhotonVision/photonvision #1415: Add current IP address to Settings Page

asid61, 2024-09-12:
**Is your feature request related to a problem? Please describe.**
I would like to know what the IP address of the device is, in case I am having trouble assigning a static IP address (this happens a lot)

**Describe the solution you'd like**
Show the current IP address in the Settings page. Should be easy.

**Describe alternatives you've considered**
Pinging photonvision.local works, but this is not well known.

**Additional context**

mcm001, 2024-09-12:
[image]

tyrius02, 2024-09-23:
There is always at least two IP addresses, loopback and others. While it is possible to filter out loopback addresses, how do you deal with two or more network adapters (wired or wireless) that are all active and have an IP address?
I have Orange Pi 5 maxes that have both wired and wireless. While I intend to disable the wireless during competition there is certainly advantages to setting up the WIFI (either as a client or as an AP) when just working with the devices and PhotonVision.

asid61, 2024-09-23:
Just give it the first IP address or list all that are gotten with your
typical IP lookup. Most people are just going to use an Orange Pi 5 because
there's no advantage to using the Max (or most other coprocessors). If it
works with the most common processors, it migt also work with the Max or
others - just needs to be tested. No need to overcomplicate it out of the
gate.

mcm001, 2024-09-24:
`CameraServerJNI.getNetworkInterfaces` (docs https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/cscore/CameraServerJNI.html#getNetworkInterfaces() ) does what you want. For example, here's the return value on my Windows computer:
```
String[13]@64
0 = "0.0.0.0"
1 = "172.21.0.1" // WSL adapter
2 = "0.0.0.0"
3 = "0.0.0.0"
4 = "25.55.200.156" // hamachi
5 = "0.0.0.0"
6 = "0.0.0.0"
7 = "0.0.0.0"
8 = "0.0.0.0"
9 = "0.0.0.0"
10 = "0.0.0.0"
11 = "0.0.0.0"
12 = "10.0.0.118" // my actual ethernet connection
```

mcm001, 2025-03-20:
OBE, its shown now

[image]

closed 2025-03-20 by mcm001