PhotonVision/photonvision #1419: Red balance does not show up on OV9782 camera

asid61, 2024-09-13:
**Describe the bug**
Arducam OV9782 has a random red balance every time I reboot the system. I cannot edit the red balance.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in an Arducam OV9782 into a Photonvision processor.

**Screenshots / Videos**
one on boot:
[image]
on another boot: 
[image]
and anotha
[image]

**Platform:**
 - Hardware Platform (ex. Raspberry Pi 4, Windows x64): Orange Pi 5
 - Network Configuration (Connection between the Radio and any devices in between, such as a Network Switch): VH-109 Radio
 - PhotonVision Version: dev-v2024.3.0-66-g06f0f7d6
 - Browser (with Version) (Chrome, Edge, Firefox, etc.): Chrome
 - Camera(s) Used: OV9782

**Additional context**

asid61, 2024-09-15:
Setting white balance via ssh + v4l2 CLI commands sory of works. The parameter is set, but the image does not change. Hue and Gain work normally. Below shows some of the v4l2 stuff.
[image]

Unplugging and replugging the camera did not fix this. The image became quite red. 
[photonvision-journalctl (7).txt](https://github.com/user-attachments/files/17004568/photonvision-journalctl.7.txt)

asid61, 2024-09-15:
[image]
red image

asid61, 2024-09-15:
v4l2-ctl commands for white balance work on the BREAD fork of PV here: https://github.com/BREAD5940/photon-image-modifier/releases/tag/Dev

Juniormunk, 2024-10-28:
Fixed by #1477

closed 2024-10-28 by Juniormunk